ct-Open-Source/tuya-convert #335: Please remove Tasmota as provided 3rd party firmware...

Jason2866, 2019-10-24:
since actual Tuya Convert does not erase flash. It is a high risk to use Tasmota as target firmware.
Tasmota needs a erased flash to work reliable, because it stores and uses settings in flash areas.
Documented here https://github.com/ct-Open-Source/tuya-convert/issues/328#issuecomment-545831626

kueblc, 2019-10-24:
The comment you linked is talking about their own bin they built with Arduino, not Tasmota. Majority of cases report success. And it is known that it is not a guaranteed process, which you must agree to before using `tuya-convert`. In any case, I'll accept a PR if you want to update the bin.

closed 2019-10-24 by kueblc

Jason2866, 2019-10-24:
This was not a finding from a self builded binary.
#328 is the flie you have in your repo. 
And every Tasmota file can fail because Tuya convert does NOT erase flash.