ct-Open-Source/tuya-convert #337: Unable to flash NEO Coolcam PIR

garynl, 2019-10-24:
I had hoped to contribute by extending the list of devices that had been flashed, but I've had problems with the  NEO Coolcam PIR. I did a quick scan to see if anyone had tried but found nothing so I had a go. I have since found a couple of posts explaining that the way this device saves power prevents the successful flashing of this device.

When I put the device into EZ config mode starts with downloading the smartconfig. The device stops flashing when the first set of dots stop. It eventually gets to 
"Device did not appear with the intermediate firmware"
(logs and output attached).

Now I seem to be in no mans land as I cannot get the tuyasmart app to detect the device in EZ config mode and I cant get my phone to stay connected to the AP the device creates when in AP Mode (slow flash). I know this is not a tuyasmart support group, but if anyone knows a way of moving forwards and getting the device converted or backwards to a usable tuya device, help would be much appreciated!

gary

[start_flash output.txt](https://github.com/ct-Open-Source/tuya-convert/files/3768332/start_flash.output.txt)
[logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/3768337/logs.zip)

kueblc, 2019-10-25:
It looks like the device rejected the schema given by the web server. Are you able to retrieve the `schema`, `schemaId`, or `productKey` from the device?

garynl, 2019-10-28:
Unfortunately I cant communicate with the device at all. Neither the
TuyaSmart app nor the SmartLife detect the device in EZ mode or AP mode.
Had a look at tuya-cli but it seems to need (some of) the data I am trying
to find.

In EZ mode the LED stops flashing consistently at 8% but nothings seems to
happen after that. In AP mode the device creates its AP (SmartLife-1E8B)
but my phone cannot connect, or stay connected, to it.

kueblc, 2019-10-29:
No, neither the `gwId` nor the `sign` are the `productKey`, `tuya-convert` does not currently have the ability to tell you what it is.

The `result` is what we are responding to the device with, we do the encrypting using the key `0000000000000000`. The `sign` is, well, a signature, that validates the encrypted data.

kueblc, 2019-10-29:
I just had a thought, maybe this device has a secondary microcontroller that wakes the ESP only when needed like #135. Worth trying to see if you can keep the ESP awake through the flashing process.

garynl, 2019-10-30:
Well tried waving frantically at the PIR while attempting an OTA flash but
no luck. I found some extensive documentation on the ESPHome github (
https://github.com/brandond/esphome-tuya_pir). Brandond used a USB-TTL tool
to put ESPHome on it and thereafter OTA could be used.

closed 2019-10-30 by garynl