ct-Open-Source/tuya-convert #338: install_prereq.sh fails when installing sslpsk

guny74, 2019-10-24:
### Before opening a new issue, please read the following

Hello,

during running  ./install_prereq.sh  I got an error

>  running build_ext
  building 'sslpsk._sslpsk' extension
  creating build/temp.linux-armv7l-3.7
  creating build/temp.linux-armv7l-3.7/sslpsk
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.7m -c sslpsk/_sslpsk.c -o build/temp.linux-armv7l-3.7/sslpsk/_sslpsk.o
  unable to execute 'arm-linux-gnueabihf-gcc': No such file or directory
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

  ----------------------------------------
>  Failed building wheel for sslpsk
  Running setup.py clean for sslpsk
Failed to build sslpsk
Installing collected packages: sslpsk
  Running setup.py install for sslpsk ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-y3e5r7i1/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-p5i5dfk2/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.7
    creating build/lib.linux-armv7l-3.7/sslpsk
    copying sslpsk/sslpsk.py -> build/lib.linux-armv7l-3.7/sslpsk
    copying sslpsk/__init__.py -> build/lib.linux-armv7l-3.7/sslpsk
    creating build/lib.linux-armv7l-3.7/sslpsk/test
    copying sslpsk/test/test_sslpsk.py -> build/lib.linux-armv7l-3.7/sslpsk/test
    copying sslpsk/test/example_server.py -> build/lib.linux-armv7l-3.7/sslpsk/test
    copying sslpsk/test/example_client.py -> build/lib.linux-armv7l-3.7/sslpsk/test
    copying sslpsk/test/__main__.py -> build/lib.linux-armv7l-3.7/sslpsk/test
    copying sslpsk/test/__init__.py -> build/lib.linux-armv7l-3.7/sslpsk/test
    running build_ext
    building 'sslpsk._sslpsk' extension
    creating build/temp.linux-armv7l-3.7
    creating build/temp.linux-armv7l-3.7/sslpsk
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.7m -c sslpsk/_sslpsk.c -o build/temp.linux-armv7l-3.7/sslpsk/_sslpsk.o
    unable to execute 'arm-linux-gnueabihf-gcc': No such file or directory
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

>Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-y3e5r7i1/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-p5i5dfk2/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-req-build-y3e5r7i1/

> 
 
Any idea how to sove the problem ?

Thank you 

Regards
  Thorsten

guny74, 2019-10-24:
Hello,

I solved it by installing  

> sudo apt-get install build-essential

Regards

closed 2019-10-24 by guny74

kueblc, 2019-10-24:
Thanks for reporting, I'll add this to our prerequisites

m0hai, 2024-02-15:
This problem is here again.
This is the output now: 
`
root@raspberrypi:/home/pi/tuya-convert# ./install_prereq.sh
Trovato:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Trovato:2 http://archive.raspberrypi.com/debian bookworm InRelease
Lettura elenco dei pacchetti... Fatto
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze... Fatto
Lettura informazioni sullo stato... Fatto
git è già alla versione più recente (1:2.39.2-1.1).
iw è già alla versione più recente (5.19-1).
dnsmasq è già alla versione più recente (2.89-1).
rfkill è già alla versione più recente (2.38.1-5).
hostapd è già alla versione più recente (2:2.10-12).
screen è già alla versione più recente (4.9.0-4).
curl è già alla versione più recente (7.88.1-10+rpi1+deb12u4).
build-essential è già alla versione più recente (12.9).
python3-pip è già alla versione più recente (23.0.1+dfsg-1+rpt1).
python3-setuptools è già alla versione più recente (66.1.1-1).
python3-wheel è già alla versione più recente (0.38.4-2).
python3-dev è già alla versione più recente (3.11.2-1).
mosquitto è già alla versione più recente (2.0.11-1.2+deb12u1).
haveged è già alla versione più recente (1.9.14-1).
net-tools è già alla versione più recente (2.10-0.1).
libssl-dev è già alla versione più recente (3.0.11-1~deb12u2+rpt1).
0 aggiornati, 0 installati, 0 da rimuovere e 0 non aggiornati.
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
`

Jbithell, 2024-04-01:
> This problem is here again. This is the output now: ` root@raspberrypi:/home/pi/tuya-convert# ./install_prereq.sh Trovato:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease Trovato:2 [...]

I managed to get around this by adding `--break-system-packages` to the pip command

```sh
sudo -H python3 -m pip install --upgrade --break-system-packages paho-mqtt tornado git+https://github.com/drbild/sslpsk.git pycryptodomex
```