ct-Open-Source/tuya-convert #339: AP showing up as an open network

Speedbreaker12, 2019-10-24:
Hey, I'm trying to flash an LSC RGB Stripe using a Raspberrypi 3b+ with a fresh install of Raspbian Buster Lite, just for this.
The AP shows up fine and I can connect to it. The problem is, it shows up as an open network, so there is no way to enter the password. So the Sending wifiPassword field is blank and the smartconfig fails. I'm using wlan0 and the adapter is free
[image]

kueblc, 2019-10-24:
Removing the password was an intentional change to streamline the process. It shouldn't interfere with the smartconfig process, it is expected that the wifiPassword field be blank. If it's not working there is probably another issue to find.

kueblc, 2019-10-24:
We're debugging support for this device in #320 so join us over there. Please share your logs too.

closed 2019-10-24 by kueblc