ct-Open-Source/tuya-convert #343: Gosund Wifi Switch - Connects to tuya-convert, but then reports "Device did not appear with the intermediate firmware"

vador98, 2019-10-25:
Everything appears to work at first - The Wifi Network is found my my smart phone, and the flashing process appears to successfully flash the switch.  However, the switch never sends a response back to the AP.  

Gosund KS-602S
Original Firmware is unknown.
Platform is Kali Linux running in VMWare using USB wifi dongle.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3773329/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3773330/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3773331/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3773332/smarthack-wifi.log)

kueblc, 2019-10-25:
Looks like there was a problem with the `psk-frontend`, or maybe several. No shared ciphers is concerning. Address was not available, maybe the socket wasn't closed from before.

Would you be able to grab a capture of the network traffic while this is happening? Something to the effect of `tcpdump -i wlan0 -w capture.pcap`

vador98, 2019-10-25:
I've not captured the network traffic before.  Hopefully this is what you're looking for!

Thank you!

[capture.zip](https://github.com/ct-Open-Source/tuya-convert/files/3773396/capture.zip)

kueblc, 2019-10-25:
The device is making calls to the right place but no one is answering the phone. Make sure `psk-front-end` is running.

vador98, 2019-10-25:
It took a little bit of playing, but I finally found a combination that worked.  After running ./start_flash.sh, while it was waiting for me to hit Enter, I manually ran psk-frontend.py from the scripts directory.  After that, I proceeded to flash the switch, which was successful.  I had to do this process 2-3 times before it appeared to take, but I've managed to successfully flash 2 switches.

Thanks for your help!

closed 2019-10-25 by vador98

ozboss, 2019-10-26:
I have the same problem.
@vador98 how do you do that?

kueblc, 2019-10-26:
@ozboss as in your `psk-frontend` shows as not running in your log? Please open a new issue and post your logs if it is not the same

ozboss, 2019-10-26:
I'm not sure but I think so:
https://github.com/ct-Open-Source/tuya-convert/issues/347

alexhk, 2019-10-29:
I flashed 8 Gosund SP111 yesterday. However, only the fist one after a fresh boot would work as planned, the others didn't. It took me a while until I figured out I had to unplug my external USB WLAN dongle and restart the VM after each Gosund SP111 was programmed. 

So in short, only after clean boot I could use start_flash.sh to program my SP111. Try rebooting your VM and unplugging your USB WLAN dongle (and connect it to the VM again if needed).

Also see my previous comment here:
https://github.com/ct-Open-Source/tuya-convert/issues/347#issuecomment-547470547