ct-Open-Source/tuya-convert #344: Unable to Flash BlitzWolf BW-LT11 Led strip

double1968, 2019-10-26:
I tried to flash the BlitzWolf BW-LT11 led strip.
Unfortunately, the operation is not successful.
I am attaching the logs hoping it can be resolved.
Thank you all.
[Logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/3774650/Logs.zip)

kueblc, 2019-10-26:
It looks like the device is rejecting the `schema`. We'll need to find the real `schema`, `schemaId` or `productKey` for this device. You can do that by capturing the registration or by using `tuyapi`.

double1968, 2019-10-26:
> It looks like the device is rejecting the `schema`. We'll need to find the real `schema`, `schemaId` or `productKey` for this device. You can do that by capturing the registration or by using `tuyapi`.

name: BW-LT11
id: 86050170bcddc239fc0a
key: 52cc115fcc0618bf

kueblc, 2019-10-27:
That's the device's `gwId` and `localKey` I presume, but we'll need the `schema` most importantly.

kueblc, 2019-10-27:
If you have it hooked up to `tuyapi`, I believe you can use `get({schema: true})`

double1968, 2019-10-27:
> get({schema: true})

Sorry but I don't know where to apply that option.
In the shell? Can you explain me.
I extracted other data with the command
tuya-cli get --all

:~$ tuya-cli get --all --id 86050170bcddc239fc0a --key 52cc115fcc0618bf
{ devId: '86050170bcddc239fc0a',
  dps: 
   { '1': false,
     '2': 'white',
     '3': 255,
     '4': 255,
     '5': '1900000000ff19',
     '6': '00ff0000000000',
     '7': 'ffff500100ff00',
     '8': 'ffff8003ff000000ff000000ff000000000000000000',
     '9': 'ffff5001ff0000',
     '10': 'ffff0505ff000000ff00ffff00ff00ff0000ff000000' } }

double1968, 2019-11-10:
With the latest version of Tuya-covert it's ok!
Many thanks.

kueblc, 2019-11-10:
Excellent, very happy to hear it

closed 2019-11-10 by kueblc