ct-Open-Source/tuya-convert #345: No AP can be created.

RestOp, 2019-10-26:
Use Raspberry with installed Openhab.

I followed the steps in manual, but after i start start_flash.sh i get no visible wifi net.

Here is thу full LOG
`Checking for network interface wlan0... Found.
Attempting to stop wpa_supplicant
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for 10.42.42.1: Address already in use
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Checking for network interface wlan0... Found.
Attempting to stop wpa_supplicant
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
RTNETLINK answers: Network is down
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for 10.42.42.1: Address already in use
Starting AP on wlan0...
Configuration file: hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:91:36 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
`
What could u advise me?

Thanks a lot!

kueblc, 2019-10-26:
> rfkill: WLAN soft blocked

Try `sudo rfkill unblock wifi; sudo rfkill unblock all`

RestOp, 2019-10-26:
Thanks!
Now i have got AP!

closed 2019-10-26 by RestOp