ct-Open-Source/tuya-convert #346: Gosund SP1 became a Brick?

RestOp, 2019-10-26:
Hello,

tried to flash it but now it do not respond. When i put it in socket, it blinks one time and thats all. I push the button, but nothing happens.
 I connected my tab with vtrust-flash network and set manual IP to 10.42.42.42 (do i need to do it?).
I started the script and got following output 

`[15:23:51] openhabian@openHABianPi:~/tuya-convert$ ./start_flash.sh
======================================================
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
./start_flash.sh: line 133:  5751 Terminated              ./smartconfig/main.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (7) Failed to connect to 10.42.42.42 port 80: Connection refused
======================================================
Getting Info from IoT-device
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
If you want to flash the included ESPurna 1.13.5 image use this command:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/espurna-base.bin
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] y
======================================================`

I understood that somthing went wrong and made to may Tab DHCP ip, it got 10.42.42.10.
I typed "yes" and then i got following 

`======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
...............
SmartConfig complete.
Resending SmartConfig Packets
...../start_flash.sh: line 133:  5822 Terminated              ./smartconfig/main.py
..................................................
Device did not appear with the intermediate firmware`

I need also to say, before flashing, i did not put any buttons on the sp1. When i put it first time in socket it was blinking blue led about 1 blink/sec - i thought it was already paiting mode, because device is new. 

So now when i put in socket it blinks one time with blue led and the nothing. I hold button about 30 sec, but nothing happends.

1. Is it dead now?
2. Сan i make it alive?
2. What have i done wrong?

Thanks a lot!

RestOp, 2019-10-26:
Here are the LOGS
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3774897/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3774898/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3774899/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3774900/smarthack-wifi.log)

RestOp, 2019-10-26:
I did not understood what has happened but started scipt one more time and got this output 
`[15:51:36] openhabian@openHABianPi:~/tuya-convert$ ./start_flash.sh
======================================================
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
./start_flash.sh: line 133:  8559 Terminated              ./smartconfig/main.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  30067      0  0:00:34  0:00:34 --:--:-- 28611
curl: Saved to filename 'firmware-a51889.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: a51889
MAC: DC:4F:22:A5:18:89
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
If you want to flash the included ESPurna 1.13.5 image use this command:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/espurna-base.bin
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] n
Exiting...
[16:06:30] openhabian@openHABianPi:~/tuya-convert$ curl http://10.42.42.42
VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: a51889
MAC: DC:4F:22:A5:18:89
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
[16:06:32] openhabian@openHABianPi:~/tuya-convert$ curl http://10.42.42.42/flash2
Device is already booting from userspace 2 (0x81000)
[16:06:54] openhabian@openHABianPi:~/tuya-convert$ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart
[16:07:06] openhabian@openHABianPi:~/tuya-convert$`

Does it means that flash was successfull???
 Now i see the SONOFF-XXXX network!

kueblc, 2019-10-26:
Yes looks like you flashed it

closed 2019-10-26 by kueblc

msandik, 2019-10-28:
Hello,

I got 3 more sockets and want to flash them also. But i did not understood, if i need to set manual ip 10.42.42.42 on the tablet  or leave dhpc and this ip will get socket?

Thanks l!

kueblc, 2019-10-29:
You should not need to do any manual assignments, for sure do not use 10.42.42.42 as it is used by the device to signal to the script that it is running the intermediate firmware