ct-Open-Source/tuya-convert #347: Cannot flash Gosund SP1

ozboss, 2019-10-26:
Cannot flash Gosund SP1.
Don't know the firmware version.
The connection to vtrust-flash network only works when is cancel with CTR+C and start the script again.
After that I can connect but without any internet beeing passed through.
I get this error when I try to flash then:

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Stopping smart config
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3775017/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3775018/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3775019/smarthack-mqtt.log)

Do you want to flash another device? [y/N]

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3775016/smarthack-psk.log)

kueblc, 2019-10-27:
From the logs it looks pretty normal. Do you want to try pulling the latest commits and retrying?

ozboss, 2019-10-27:
Still the same.
When I start the script I have to cancel it and start again so that I'm able to connect to the hotspot.
When I run the script then it seems to connect to the SP1.
The light of the SP1 stops blinking and turns red while the script tries to resend SmartConfig Packets.
Here the new logs:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3776217/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3776218/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3776219/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3776220/smarthack-wifi.log)

scottsuine, 2019-10-29:
I am getting the same error. Have tried to flash two different devices and end up with the same outcome.

alexhk, 2019-10-29:
For what it's worth, I flashed 8 Gosund SP111 yesterday (pulled the latest commit prior to that).

I  am using the vmware KaliLinux image and an external WLAN adapter.

Every time after a flash was done, I had to restart the VM and unplug the WLAN adapter to rest it, it was somehow stuck. If I wouldn't unplug and replug the external WLAN adapter (I assume restarting the whole machine would have done the same), the WLAN adapter would not really work inside the VM - for instance I couldn't scan available networks before starting tuya-convert/start_flash.sh.

Long story short: it may work with your SP1's if you do a full reset of the machine and try to run start_flash.sh after a fresh boot.

It took me a while until I fugured that out. Previously I always had flashed my SP111 without problems. Something must have been changed on recent commits.

ozboss, 2019-10-29:
> 
> 
> For what it's worth, I flashed 8 Gosund SP111 yesterday (pulled the latest commit prior to that).
> 
> I am using the vmware Kali Linux image and an external USB WLAN adapter.
> 
> Every time after a flash was done, I had to restart the VM and unplug the WLAN adapter to completely reset it, it was somehow stuck without that. If I wouldn't unplug the external WLAN adapter (I assume restarting the whole machine would have done the same), the WLAN adapter would not really work inside the VM - for instance I couldn't scan available networks before starting tuya-convert/start_flash.sh.
> 
> Long story short: it may work with your SP1's if you do a full reset of the machine and try to run start_flash.sh after a fresh boot.
> 
> It took me a while until I figured that out. Previously I always had flashed my SP111 without problems. Something must have been changed on recent commits.

Thanks for the tip but I did it with a Raspberry Pi 3 and Raspbian Light so this probably doesn't apply.

ozboss, 2019-10-29:
Btw I bought the SP1's when the new Tuya fix was just released so I couldn't flash them.
So this is why I expected it to work fine now with the new tuya-convert...

ozboss, 2019-10-29:
Ok not when it was just released but in August.
So this should work shouldn't it?

tameike000, 2019-11-08:
+1 
Same issue here. Just bought fresh devices at Amazon and using a RPI.

kueblc, 2019-11-08:
@tameike000 can you post your logs?

tameike000, 2019-11-08:
> @tameike000 can you post your logs?

done.

kueblc, 2019-11-08:
@tameike000 not the same issue. See in your web log:
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```

tameike000, 2019-11-08:
@kueblc Does this mean Tuya changed the hardware?

kueblc, 2019-11-08:
Yes, I know that they're using the RTL8710 in a few of their modules now. They're pin out compatible with the ESP so manufacturers can start using the new module without changing any other hardware.

kueblc, 2019-11-08:
See #269

kueblc, 2019-11-10:
@ozboss did you retry? Is this still an issue?

ozboss, 2019-11-10:
I was reading about bricked devices because the firmware was too big, so I thought I will wait a little :D but the issue seems to be solved 👍 
Will try in a bit.

ozboss, 2019-11-10:
@kueblc bro you are amazing :) Did the first plug and it worked flawelessly. Three more to go.
Thank you so much :)

closed 2019-11-10 by ozboss

kueblc, 2019-11-10:
Awesome, really happy to hear it