ct-Open-Source/tuya-convert #348: Failed to connect No route to host

pudah, 2019-10-26:
I am trying to use tuya-convert to flash some Novostella smart bulbs.

I am using Kali linux on a raspberry pi2 with a usb wifi dongle.

I successfully installed the intermediate firmware on all three in succession.

After doing that, I went to check each bulb with curl using:

# curl http://10.42.42.42

The curl command retreived the buld info for 2 of the 3 bulbs. But one got the following error:

curl: (7) Failed to connect to 10.42.42.42 port 80: No route to host

For the two that successfully pulled the bulb info, I was able to flash tasmota.

But I continue to get the No root to host on the third bulb.

This is the info from the intermediate flash for the bulb:

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
...............
SmartConfig complete.
Resending SmartConfig Packets
...............................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  29475      0  0:00:35  0:00:35 --:--:-- 28733
curl: Saved to filename 'firmware-a1140d.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: a1140d
MAC: BC:DD:C2:A1:14:0D
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================

Any suggestions on how to solve this problem? Or have I bricked the bulb?

pudah, 2019-10-27:
can you provide a little more info on how I would go about doing this?

kueblc, 2019-10-27:
Let me see if I follow you, you flashed three devices, and the log you posted is from one of the two that was successful, but one of the three did not get that far. Is that right?

Since the other two worked, it may have just been a fluke. Try restarting the process but instead of putting the device into pairing mode and connecting a phone, just unplug and plug the device back in.

pudah, 2019-10-27:
Not exactly. All three bulb got thru the intermediate flash. All three produced summary information like that above.  What is above is from the bulb that got past the intermediate flash, but which does not respond when I attempt the final flash.

I too find it perplexing that I was able to flash two of the three bulbs but not this other one.

I have already tried several times to do the flash. I unplugged the bulb and retried 2 or 3 times. I rebooted the rpi and tried again. Every time I get the same Failed to connect No route to host error message.

kueblc, 2019-11-17:
Any news on this @pudah? Have you tried plugging the device back in and looking for a `vtrust-recovery` SSID?

kueblc, 2019-11-21:
Closing due to inactivity, feel free to reopen if this is still an issue

closed 2019-11-21 by kueblc