ct-Open-Source/tuya-convert #349: Can't obtain IP address from vtrust-flash

CervDotBe, 2019-10-26:
Hi all,

I am using a raspberry pi with tuya convert on it. I am not able to connect to the vtrust-flash AP properly, it gives no IP address. I get some un clear dnsmasq error in my smarthack-wifi.log:

> Starting AP on wlan0...
Configuration file: hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:27:88:11 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 18:01:f1:1e:87:b8
wlan0: AP-STA-DISCONNECTED 18:01:f1:1e:87:b8
wlan0: AP-STA-CONNECTED 18:01:f1:1e:87:b8
wlan0: AP-STA-DISCONNECTED 18:01:f1:1e:87:b8
wlan0: AP-STA-CONNECTED 18:01:f1:1e:87:b8
Checking for network interface wlan0... Found.
Attempting to stop wpa_supplicant
Configuring AP interface...
Starting DNSMASQ server...

**dnsmasq: failed to create listening socket for 10.42.42.1: Address already in use**

Thanks in advance!

kueblc, 2019-10-27:
You'll need to close the process that's conflicting with `dnsmasq` on port 53.

CervDotBe, 2019-10-27:
How can I identify the process and kill it? I am no Linux expert. :( I started from a clean Raspbian Buster Lite install so I am confused why there is a conflicting process.

PavelGorbanj, 2019-10-27:
@CervDotBe you may want to start with https://github.com/ct-Open-Source/tuya-convert/issues/157 in general, it's useful to search for similar issues first.

CervDotBe, 2019-10-28:
I did for hours. Issue #157 I already took a look at. No fix.

kueblc, 2019-10-28:
My guess would be `systemd-resolved`. To close that try `sudo service systemd-resolved stop`.

If not try `lsof -Pni tcp:53 -s TCP:LISTEN -t` and then `kill` the PID it gives

PavelGorbanj, 2019-10-28:
just in case, another dumb idea is to check if host IP overlaps with one
used by AP/vtrust-flash.

Regards,
  Pavel

CervDotBe, 2019-10-29:
> My guess would be `systemd-resolved`. To close that try `sudo service systemd-resolved stop`.
> 
> If not try `lsof -Pni tcp:53 -s TCP:LISTEN -t` and then `kill` the PID it gives

Life saver :)

Thanks!

closed 2019-10-29 by kueblc