ct-Open-Source/tuya-convert #353: Convert bricks auvision URC-150

wmchris, 2019-10-27:
Env: Raspi 4b 4GB

Flashing started but didnt find the device
```
SmartConfig complete.
Resending SmartConfig Packets
.....................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
```

device has been found and regarding the logs, the upgrade.bin has been requested from http. 
Now the device is bricked, no lights are flashing anymore. Seems like i need to solder it up.

wmchris, 2019-10-27:
More details:
```
POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=04314116cc50e3463c5e&t=1572186306&sign=1450cb3bc8e51f1ca46ef1ae786774c3^M
Host: 10.42.42.1^M
Ty-Ua: HW^M
Connection: keep-alive^M
Content-Type: application/x-www-form-urlencoded; charset=UTF-8^M
Content-Length: 69^M
^M
payload {"upgradeStatus":2}^M
Answer s.gw.upgrade.updatestatus^M
reply {"t":1572186306,"e":false,"success":true}^M
^[[32m[I 191027 14:25:06 web:2246]^[[m^O 200 POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=04314116cc50e3463c5e&t=1572186306&sign=1450cb3bc8e51f1ca46ef1ae786774c3 (10.42.42.13) 50.74ms^M
^[[32m[I 191027 14:25:08 web:2246]^[[m^O 206 GET /files/upgrade.bin (10.42.42.13) 10.14ms^M
encoding {"data":{"gwId":"04314116cc50e3463c5e"},"protocol":15,"s":1523715,"t":1572186312} using protocol 2.1^M
b'2.18b65bf5ba4bb0146t5inCMDHpcJKQpxfOe2RbTyiJGfrU9NhHEb8LAqfQW9Q2YdxxaLPDRsdgcuKCq/ZJsm5uMafEvnzLLz/1MQ4v+lxiVFX0tLC5gUeCaYQh9rdg4SXZL9kLEC9aqKXAOwI'^M
^[[32m[I 191027 14:25:15 web:2246]^[[m^O 206 GET /files/upgrade.bin (10.42.42.13) 7405.38ms^M
```

kueblc, 2019-10-27:
In a majority of cases, the device is not actually bricked, but just stuck waiting for the next step. It looks like your device successfully flashed the intermediate firmware.

Unplug the device, restart `start_flash`, but instead of hooking up a phone and putting the device in pairing mode, just plug the device back in.

wmchris, 2019-10-31:
Sadly didn’t work. It was bricked as there was no WiFi connection.

kueblc, 2019-10-31:
If you decide to open up and flash over wire, please make a backup so we can debug what went wrong

stevenjev, 2019-11-05:
I tried the advice to `start_flash` and then plugging the device in multiple times (10+) with no luck. However, I then tried holding the power button down when plugging in and instantly the device appeared. This was with a different device (ZooZee SA102) but might be something to try.

kueblc, 2019-11-10:
@wmchris did you replace any of the files in your `tuya-convert` installation? Have you checked for a `vtrust-recovery` SSID while the device is plugged in?

kueblc, 2019-11-17:
Closing due to lack of response, please reopen if this is still an issue

closed 2019-11-17 by kueblc

pnlinh-it, 2020-01-13:
@kueblc I also have same issue. Log files:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4054677/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4054678/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4054679/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4054680/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4054681/smarthack-wifi.log)

kueblc, 2020-01-22:
Hi @pnlinh-it, I do not see the relation between your issue and this one. I do not see any ESP devices connected in your wifi log. Let's move the conversation to a new issue if you need further assistance.

pnlinh-it, 2020-01-23:
> Hi @pnlinh-it, I do not see the relation between your issue and this one. I do not see any ESP devices connected in your wifi log. Let's move the conversation to a new issue if you need further assistance.

@kueblc  Thanks for supporting, I tried many times but it failed. So I decide to use wire 🙇.