ct-Open-Source/tuya-convert #354: Cannot Flash 'Arlec Smart Led Globe RGB'

lowey71, 2019-10-29:
Have disassembled one, so can dump flash at some point.  tywe3l based

Believe it is rejecting schema

tuyapi-cli dump:

{ devId: '55866400a4cf12ab8449',
  dps:
   { '20': true,
     '21': 'white',
     '22': 500,
     '23': 1000,
     '24': '000003e803e8',
     '25': '000e0d00002e03e802cc00000000',
     '26': 0 } }

How can I use this information to assist in flashing?

kueblc, 2019-10-29:
`schema` is defined [here](https://github.com/ct-Open-Source/tuya-convert/blob/3c246a9511f063b491a89109ef9861a742d8812e/scripts/fake-registration-server.py#L153), it will need to be updated to include dps 20-26

Alfiegerner, 2019-10-30:
I have a similar situation (Kogan Color Temp bulb), with dps starting at 20.

Any chance you could tell us what change is needed here - I tried setting id to to 20, also schema_key_count to 20 in the line you linked to above:
                {"mode":"rw","property":{"type":"bool"},"id":1,"type":"obj"}] * schema_key_count)
Neither worked, but then I may have ran into other issues.

Thanks

kueblc, 2019-10-30:
Please share your web log @Alfiegerner

Alfiegerner, 2019-10-30:
Here you go: [smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3791437/smarthack-web.log)

Alfiegerner, 2019-10-30:
Got it working!  Changing schema_key_count to 20 worked when I reverted back to master (think I'd screwed up my dev branch trying to make progress).

Just for clarity: I changed this line:
                {"mode":"rw","property":{"type":"bool"},"id":1,"type":"obj"}] * schema_key_count)
to 
                {"mode":"rw","property":{"type":"bool"},"id":1,"type":"obj"}] * 20)

codyc1515, 2019-10-30:
Also facing a similar issue with the Deta branded products (same as or similar to Arlec brand) from Bunnings.

Alfiegerner, 2019-10-30:
> Also facing a similar issue with the Deta branded products (same as or similar to Arlec brand) from Bunnings.

Try details from tuyapi-cli dump to see what DPS number the device starts with, then make the change to the fake_registrations_server.py where I've replaced schema_key_count with 20 (my Kogan bulbs had attributes starting from DPS 20).

lowey71, 2019-10-31:
As above, changing schema_key_count to 20 has allowed the device to flash fine.  Thanks @Alfiegerner and @kueblc !!

closed 2019-10-31 by lowey71

kueblc, 2019-10-31:
Awesome, great to hear. Would you @Alfiegerner and @lowey71 share your web log and firmware backups? I would like to do some testing before including the fix in `master`.

Alfiegerner, 2019-11-01:
> Awesome, great to hear. Would you @Alfiegerner and @lowey71 share your web log and firmware backups? I would like to do some testing before including the fix in `master`.

@kueblc , Attached, thanks very much for your help.
[tuya_dps_20.zip](https://github.com/ct-Open-Source/tuya-convert/files/3799221/tuya_dps_20.zip)

lowey71, 2019-11-02:
Hope this helps!

[firmware-ab8449.zip](https://github.com/ct-Open-Source/tuya-convert/files/3800456/firmware-ab8449.zip)

commit 4aa350c1 (2019-11-07): Modify extended schema to cover more cases, fix #354 #363 #369 #372