ct-Open-Source/tuya-convert #355: Cannot Flash LSC Smart Door Sensor (action)

ASPI259, 2019-10-29:
i have already flashed some lamps with Tuya-Convert, works fine, i will also add them to the wiki.

This time I tried the whole thing with the "LSC Smart Connect Smart Door Sensor" from the Action Store.

I get 10 times in a row:

SmartConfig complete.
Resending SmartConfig Packets
.............................

and then it just comes:
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder

but I can't see anything unusual in the logs.

Does anyone else have a problem like that?

Translated with www.DeepL.com/Translator from Germany

kueblc, 2019-10-29:
The device is not staying online. Devices like this use a secondary microcontroller that wakes up the ESP only when there is data to send. Duplicate of #135, try keeping the ESP active as the last comment suggests.

closed 2019-10-29 by kueblc

ASPI259, 2019-10-29:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3784635/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3784636/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3784637/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3784638/smarthack-wifi.log)

I did i like in the last Comment of #135 but it doesnt work.