ct-Open-Source/tuya-convert #356: It works! But not correct bvehavior!

RestOp, 2019-10-29:
Hello,

Thanks one more time so much for this method! It works!
I flashed 4 Gosund S1 with it, but always with faults.

I need to start 2-3 times script to get it flashed.
Please look for it:

[18:01:27] openhabian@openHABianPi:~/tuya-convert$ ./start_flash.sh
======================================================
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID vtrust-flash
Sending wifiPassword
Sending token 00000000
Sending secret 0101
................
SmartConfig complete.
Resending SmartConfig Packets
......./start_flash.sh: line 133: 3051 Terminated ./smartconfig/main.py
...............................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to flash another device? [y/N] я
Exiting...
[18:03:46] openhabian@openHABianPi:~/tuya-convert$ ./start_flash.sh
======================================================
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
./start_flash.sh: line 133: 3723 Terminated ./smartconfig/main.py
  % Total % Received % Xferd Average Speed Time Time Time Current
                                 Dload Upload Total Spent Left Speed
100 1024k 100 1024k 0 0 29856 0 0:00:35 0:00:35 --:--:-- 28472
curl: Saved to filename 'firmware-6f9352.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 6f9352
MAC: 84:0D:8E:6F:93:52
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
If you want to flash the included ESPurna 1.13.5 image use this command:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/espurna-base.bin
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] т
Exiting...
[18:04:43] openhabian@openHABianPi:~/tuya-convert$ curl http://10.42.42.42
VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 6f9352
MAC: 84:0D:8E:6F:93:52
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
[18:04:46] openhabian@openHABianPi:~/tuya-convert$ curl http://10.42.42.42/flash2
Device is already booting from userspace 2 (0x81000)
[18:04:48] openhabian@openHABianPi:~/tuya-convert$ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart
[18:04:53] openhabian@openHABianPi:~/tuya-convert$

kueblc, 2019-10-29:
Can you explain what the problem is? I don't see what behavior you're referring to

RestOp, 2019-10-29:
I flashed 4 socket and everytime behavior is not the same.

I start ./start_flash.sh first time and get 
`
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets 
`
then start one more time  ./start_flash.sh
i get 

`................
SmartConfig complete.
Resending SmartConfig Packets
......./start_flash.sh: line 133: 3051 Terminated ./smartconfig/main.py
...............................................
Device did not appear with the intermediate firmware`
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets 
`
and again error like `......./start_flash.sh: line 133: XXX Terminated ./smartconfig/main.py`
and `Device did not appear with the intermediate firmware``

Then device makes click like relays clicks and no LED active. I need to push (not seeing if LED blink fast or not because LED do not work) it long to get gevice in pairing mode.

Then it goes how its must.

kueblc, 2019-10-29:
Are you using the latest commit?

If you need to restart `start_flash` you do not need to entering pairing mode again for the same device.

If the LED does not light it's probably already running the intermediate firmware.

kueblc, 2019-10-29:
Closing as there is not actually a problem since flashing was successful

closed 2019-10-29 by kueblc

RestOp, 2019-10-29:
> Are you using the latest commit?
> Closing as there is not actually a problem since flashing was successful
Yes! Last one!
I have made it successfully, but it not clear for everyone i afraid(((

darkfader, 2020-03-28:
I got an error too. But I got the LED to stay on continuously and tried again.
Now it went further but the backup failed. LED went off too.
```
./start_flash.sh: line 129: 18254 Terminated              ./smartconfig/main.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:01:30 --:--:--     0
curl: (28) Connection timed out after 90001 milliseconds
Could not fetch a complete backup
Do you want to continue anyway? [y/N] y
```
After pressing button again, I continued from this point and it started flashing anyway and succeeded!