ct-Open-Source/tuya-convert #358: vtrust-flash didn't show up

chupachupbum, 2019-10-30:
I'm using Ubuntu 16.04 32 bit, I followed all the steps, did use sudo rfkill unblock wifi as I saw I had the same problem in another issue

Here is the most recent log:

Checking for network interface wlp3s0... Found.
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
Starting DNSMASQ server...
Starting AP on wlp3s0...
hostapd: invalid option -- 'i'
hostapd v2.4
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2015, Jouni Malinen <j@w1.fi> and contributors

usage: hostapd [-hdBKtv] [-P <PID file>] [-e <entropy file>] \
         [-g <global ctrl_iface>] [-G <group>] \
         <configuration file(s)>

options:
   -h   show this usage
   -d   show more debug messages (-dd for even more)
   -B   run daemon in the background
   -e   entropy file
   -g   global control interface path
   -G   group for control interfaces
   -P   PID file
   -K   include key data in debug messages
   -f   log output to debug file instead of stdout
   -T = record to Linux tracing in addition to logging
        (records all messages regardless of debug verbosity)
   -t   include timestamps in some debug messages
   -v   show hostapd version
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
Could not start server on port 80
[Errno 99] Cannot assign requested address
Traceback (most recent call last):
  File "./psk-frontend.py", line 105, in <module>
    main()
  File "./psk-frontend.py", line 91, in main
    proxies = [PskFrontend(gateway, 443, gateway, 80), PskFrontend(gateway, 8886, gateway, 1883)]
  File "./psk-frontend.py", line 42, in __init__
    self.server_sock = listener(listening_host, listening_port)
  File "./psk-frontend.py", line 16, in listener
    sock.bind((host, port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address

chupachupbum, 2019-10-31:
@kueblc It doesn't seem like an update is available. Does that mean it can only work with hostapd v2.6 or higher?

kueblc, 2019-10-31:
I'm not sure what the minimum version might be. The `-i` flag your `hostapd` is complaining about is to set the interface. As a workaround, you can edit `hostapd.conf` to specify your interface ie `interface=wlan0` and remove `-i $WLAN` from `setup_ap.sh`.

chupachupbum, 2019-11-02:
I decided to dump that laptop and moved on with a better one, so I didn't have a chance to try your solution. Might close for now.

closed 2019-11-02 by chupachupbum