ct-Open-Source/tuya-convert #359: Listening on Port 80 * Failed to flash

lvibe, 2019-10-31:
I am having an issue with flashing a ce smart dimmer... I was previously able to flash a ce smart plug however i can not get a dimmer switch to flash. Reviewing my logs i do notice on smarthack-web that i am getting connection refused errors on port 80. When i go to flash i just get stuck in a Smart config complete loop:
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] y

I have been scouring forums for a resolution but now luck.

[smarthack-mqtt.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/3791858/smarthack-mqtt.log.txt)
[smarthack-psk.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/3791859/smarthack-psk.log.txt)
[smarthack-web.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/3791860/smarthack-web.log.txt)
[smarthack-wifi.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/3791861/smarthack-wifi.log.txt)

kueblc, 2019-10-31:
Related https://github.com/ct-Open-Source/tuya-convert/issues/333#issuecomment-546555117

closed 2019-11-02 by kueblc

lvibe, 2019-11-04:
I was able to get it working consistently now with sleep 7 lines added.

kueblc, 2019-11-04:
Great, the fix has been included now.