ct-Open-Source/tuya-convert #360: WAP settings - stuck.

Just-another-pleb, 2019-10-31:
Sorry, but I am missing something in how you describe what to do.
I have to rename my machine (Ubuntu) to vtrush-flash and set the password for the WAP.
And I also have to set the IP address to 10.24.24.24 - or so I read.
I try that, but also need to set up a netmask.   255.255.255.0
I see the  WAP from my phone but as soon as I try to connect it all just falls over.
The WAP dies.
It "worked" ok - getting IP - before but when I suspected/realised the IP also needs to be set, and did that, it just kills the WAP when the phone tries to get an IP address.

Yes, probably beyond the scope of this update, but I'm stupid.
I am not getting how this is supposed to work.

The name of the WAP the board gives is not really explained, but I guess that isn't a problem.
Oh, it also has no password.   So it is an open WAP.

I have an INTEL I7 NUC running Ubuntu 18.04 LTS.  The WAP device is wlp58s0  and I edited the config.txt file to suit.

Could you help me understand why the NUC's WAP is not working and clarify if the IP address is really needed/important for it to work.

kueblc, 2019-10-31:
You shouldn't need to perform any of these steps manually, the script `setup_ap` does all this when called from `start_flash`. Just set `WLAN=wlp58s0` in your `config.txt` and run `start_flash`.

Just-another-pleb, 2019-10-31:
Thanks.

So, given what you said:
That would/could explain why I lose the network icon when the script is run.

However, it doesn't come back and I see the script trying to send data.

I'll keep digging.

Appreciated.

Mind if I leave this open for now?

kueblc, 2019-10-31:
> That would/could explain why I lose the network icon when the script is run.

Yes, you won't be connected as a station while running the `vtrust-flash` AP.

> However, it doesn't come back and I see the script trying to send data.

When you're done, run `stop_flash`

> But "setup_ap.sh" is in the scripts directory.
> Which is not where I "am* when I run the script. I'm in the parent directory.

You'll see that earlier in the script we change directory with `pushd`

> I am trying to do everything from one machine.

That shouldn't be a problem, it should be run from one machine. You'll only need another device to connect to the `vtrust-flash` AP in order to complete the pairing procedure.

kueblc, 2019-11-03:
@Just-another-pleb have you resolved your issue?

Just-another-pleb, 2019-11-03:
(Sorry been away and now playing "catch up".)

Not really.

It would seem I need two machines to do this and I am just not ready for that level of complication.

I shall get in touch with someone I know who may be able to help me.  He has done a couple of these before I believe.

Sorry to have bothered you.   (Not blaming you.)

kueblc, 2019-11-03:
No problem. Not sure I understand why you would need two machines for this, but let me know if I can help any further. Good luck!

Just-another-pleb, 2019-11-07:
Sorry to go on.  (And bother you)

I got it flashed and it is here, in front of me.

I am now a bit confused with how to set it up now.
I open the CONFIGURE part then CONFIGURE MODULE.

[image]

So I am guessing I have to "cross connect" things so the 3 relays are assigned to different GPIO pins.

Yeah, but I am not getting what is what in the picture.

(And I don't get/know/understand what to send in MQTT)

I've got the name - set to BAUHN-1- which is all well and good.  But I am stuck beyond that.

I've opened the terminal and send commands like   power1 on  and power1 off
Nothing happens.

Pressing the TOGGLE on the web page, I see the MQTT message.   But no relays change state.
Mind pointing me in the right direction?

Just-another-pleb, 2019-11-07:
Update:
I got it working, but not as nice as I would like.

This is the configuration screen:

[image]

You will see that I have the relays as:  1, 2, 4, 3, 5, 6.
I'll get to that.

This is the "gui" screen I see:

[image]

The only buttons that work are:  3, 4, and 5.

But those buttons are for relays 1 2 and 3 on the board.   (It only has 3 relays.)

If I make the first two "none" the other buttons don't work.

The board is (in sequential order) are:  relay 4, relay 3 and 6.

I am not able to get only 3 buttons on the screen controlling relay 4, 3, 6.

If I have anything less than what I show, the buttons don't work.

As it is:  Button 1, 2, and 6 do nothing.

Can you explain what I am missing, or is this a software problem?

kueblc, 2019-11-07:
Glad you figured out your flashing issue! Tasmota is a separate project, you'll need to seek help there for post installation issues.

closed 2019-11-07 by kueblc