ct-Open-Source/tuya-convert #361: Unable to flash Zoozee SE131 smart wlan plugs

scholzmichael, 2019-10-31:
Trying to flash Zoozee SE131 without any luck, someone have an idea? 

https://www.amazon.de/gp/product/B07J2CT8WL/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

but only getting the following:

Starting pairing procedure in screen
RTNETLINK answers: File exists
~/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
..........................................................................................................................

redone the pi several times  with strech and tried ct-Open-Source/tuya-convert as well as M4dmartig4n/tuya-convert , here are my logs:

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3793402/smarthack-psk.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3793403/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3793404/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3793405/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3793406/smarthack-mqtt.log)

kueblc, 2019-10-31:
This device does not use an ESP82xx and therefore you can't install custom firmware like Tasmota which is built for the ESP82xx, OTA or otherwise. Unfortunately I'm not aware of any custom firmware for the Realtek chip your device is running.

closed 2019-10-31 by kueblc

JamesT42, 2019-11-15:
@kueblc the device type is listed in the tasmota wiki as supported. Where do you see that the manufacturer has now switched to realtek?

kueblc, 2019-11-15:
The behavior of the device as reported by the log

GoaMetz, 2019-11-19:
Heho,

i opend up a recently bought one and can confirm that there is a module with Realtek Chip mounted.
The module is named SP21, chip beeing "W302" (probably a RTL8710BN).

Jpsy, 2020-01-11:
From https://templates.blakadder.com/zoozee_SE131.html : 

> WARNING: There is a new version with a T102/WR2 WiFi module in it. This uses a RTL8710BN chip and can not be converted to Tasmota. Make sure that you get a plug with TYWE2S WiFi module based on ESP8266.