ct-Open-Source/tuya-convert #362: Device bricked?

FaserF, 2019-10-31:
Hi,

I have just tried to flash a device and I got a success report, but after the curl http://10.42.42.42/flash3 command the device seemed to restart and now no LED is visable, the device cant be pinged and I cant reache it in any way. Is it now completely bricked?

Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
.......................................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  78645      0  0:00:13  0:00:13 --:--:--   97k
curl: Saved to filename 'firmware-932c20.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 932c20
MAC: 60:01:94:93:2C:20
BootVersion: 4
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
If you want to flash the included ESPurna 1.13.5 image use this command:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/espurna-base.bin
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] N
Exiting...
root@DietPi:/home/dietpi/tuya-convert# curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart

FaserF, 2019-10-31:
Now I have seen it hasnt flashed sonoff-basic.bin, it flashed thirdparty.bin instead. I think my thirdparty.bin file was too large. Is there any way to recover it?

kueblc, 2019-10-31:
Could you please share your web log?

FaserF, 2019-11-01:
> Also check your wifi log to see if it reconnected.
> 
> > I think my thirdparty.bin file was too large
> 
> Did you replace `thirdparty.bin`? The included `thirdparty.bin` should be fine.

yes, I replaced it.
As the device currently cant be reached by web or wifi I dont know how to get a log.
What exactly has the device done if the bin file was too large? Is my switch now completely with no firmware on it?

kueblc, 2019-11-01:
The web and wifi logs are in your `tuya-convert/scripts` directory.

kueblc, 2019-11-01:
Could you attach your `thirdparty.bin` and the `firmware-932c20.bin`?

FaserF, 2019-11-01:
@kueblc thanks already. 
Here is the firmware: https://drive.google.com/drive/folders/10hkVnl_RgnoZMTR9ggpdElOeffScZe8y?usp=sharing
and my thirdparty.bin was sonoff.bin from official tosmata.

Log files are gone, as I have flashed another device afterwards successfully (after switching my thirdparty.bin from sonoff.bin to sonoff-basic.bin).

kueblc, 2019-11-01:
If you're talking about the most recent release (6.7.1) it looks like `sonoff.bin` is 564 KB. The intermediate firmware should have rejected this on the basis that it is too large. It would make 3 attempts and then restart.

FaserF, 2019-11-01:
@kueblc I have already tried restarting the switch multiple times. It seems completely dead, no LED, pressing the button does nothing and there is no wifi sending from the device.
Or do you mean I will have to start the Tuya convert wifi and it will auto connect with that again?

kueblc, 2019-11-01:
Yes you'll need to run `start_flash.sh` so the `vtrust-flash` AP is available. Then you should be able to `curl http://10.42.42.42` and proceed to flashing.

FaserF, 2019-11-01:
Seems not to work, the device wont be detected for the Smartconfig Packets

Baeen, 2019-11-03:
I had the same issue after replacing the thirdparty.bin with sonoff.bin from Sonoff-Tasmota 6.7.1.Allison.
Reran the flashing cycle again, but the bulb didn't show up as connecting to the AP and it definately wasn't running a sonoff-XXXX AP.
Tried power cycling the bulb a few times and no response.

FaserF, 2019-11-03:
So it is dead with no hope of fixing it? :(

kueblc, 2019-11-03:
@FaserF @Baeen did you happen to try `/undo`? It looks like this might be causing problems if the SDK version is different between the stock and intermediate firmware.

nahcir, 2019-11-03:
I was stupid to do EXACTLY the same thing.

`======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
..............................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  29966      0  0:00:34  0:00:34 --:--:-- 28965
curl: Saved to filename 'firmware-23685e.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 23685e
MAC: A0:20:A6:23:68:5E
BootVersion: 4
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440e0
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
If you want to flash the included ESPurna 1.13.5 image use this command:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/espurna-base.bin
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] N
Exiting...
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/sonoff-DE.bin
Device should flash http://10.42.42.1/files/sonoff-DE.bin and restart`

My RGB-LED however does light up in white for 6-7 seconds, turns off for a second, and then on again. I can't seem to be able to put it back in EZ mode and no AP opened by it seems to be present. I messed up, didn't I ^^?

kueblc, 2019-11-03:
Unfortunately yes, it is looking like the safety mechanism to avoid too large firmware uploads is broken in the latest release.

I'll be working on a fix with high priority but unfortunately the only way to recover devices affected by this may be to open the device and flash it over serial.

kueblc, 2019-11-05:
@FaserF @Baeen @nahcir can any of you confirm that this occurred after `/undo`? Or does it occur without `/undo`?

nahcir, 2019-11-05:
It happened without /undo to me.

kueblc, 2019-11-05:
Would you kindly share your `firmware-23685e.bin` backup?

nahcir, 2019-11-05:
Sure, of course.

https://cloud.richan.de/index.php/s/zZfeGZEntr2j4jM

kueblc, 2019-11-05:
Thank you! I was able to reproduce the issue. I'll plan on releasing a fix by the end of the week.

nahcir, 2019-11-05:
You're very welcome, thanks for looking into this.

FaserF, 2019-11-07:
> @FaserF @Baeen @nahcir can any of you confirm that this occurred after `/undo`? Or does it occur without `/undo`?

I haven't used /undo (as I couldn't reach the device anymore after flashing the wrong bin file)

kueblc, 2019-11-10:
Thanks again for everyone's help in this thread. This issue should be avoided with the latest release. Now if flashing fails for any reason, including too large firmware, it will report the error and abort early, giving you an opportunity to recover.

closed 2019-11-10 by kueblc

d4icon, 2020-01-21:
Hi,
I had the same problem as you after perfectly flashing 3 devices.
Now I have tried to update the 4th and the same thing happened here.
I have read the thread, but I still don't understand the cause of the cause.
I have done the same steps as for the previous 3.

What has happened? and how to avoid it for next time?
Thank you

kueblc, 2020-01-22:
Hi @d4icon, please open a new issue with your logs attached so I can further triage your problem.