ct-Open-Source/tuya-convert #363: Unable to get pas the SmartConfig complete step for Tuya Bulbs

uatuba, 2019-11-01:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2019-11-01:
Can you share your web log? You can attach the log as a file

Homer-Sim, 2019-11-03:
Same behaviour with teckin socket SP21
"Device did not appear with intermediate firmware"

Thanks for your help

kueblc, 2019-11-03:
@Homer-Sim please open a new issue and post your logs

uatuba, 2019-11-03:
Sorry for the delay. I worked 7:00 am CST yesterday until 3:00 am CST today, so I'm just now posting.

Here's the web log:

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3801845/smarthack-web.log)

kueblc, 2019-11-03:
Seems related to #354

garnern2, 2019-11-03:
That solution worked! Thanks so much.

Homer-Sim, 2019-11-03:
> 
> 
> That solution worked! Thanks so much.

What exactly did you do?

uatuba, 2019-11-03:
Replaced the scheme_key_count with the number 20. That’s it.

kueblc, 2019-11-03:
@Homer-Sim we should check first that you have the same problem, since you haven't posted any logs there is no way for me to tell what the actual issue is.

Homer-Sim, 2019-11-04:
sorry for delay but I was at work ;) 

first I tried with original settings ... error
then I changed fake-registration-server.py with scheme_key_count = 20
rerun but still receive the error
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3804851/smarthack-web.log)

garnern2, 2019-11-04:
You *replace* schema_key_count with 20. No one said anything about = 20...

kueblc, 2019-11-04:
@Homer-Sim I could have told you that had you shared your logs, it is not related to this issue at all. Your problem is stated clearly in the first few lines of your web log:
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```

Thanks for reporting back @uatuba @garnern2, we'll get this fix integrated ASAP!

closed 2019-11-04 by kueblc