ct-Open-Source/tuya-convert #365: sslpsk error

biotecov, 2019-11-01:
Hello,
I am trying to flash a curtain smart switch that contains a tywe3s without opening the device.

I flashed successfully another unit using serial, but it is a BIG pain in this model for purely physical reasons (complicated desoldering), and I would like to install several such switches at home, so I have a big hope that tuya-convert will be my salvation.

However I did not succeed. I know the device connects to the fake AP but then fails to get the intermediate firmware. I *think" the problem is on my sslpsk because smarthack-psk.log is full of messages like "could not establish sslpsk socket: <class 'ssl.SSLError'> returned a result with an error set", or "[SSL: NO_SHARED_CIPHER]".

I cannot figure-out what is wrong.

I am no using any debian-derivative, but a current 64-bit slackware, with python3, plus hostapd and mosquitto installed manually (no problem with those). 

I attach the log files.
Thanks for this wonderful approach.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3798309/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3798310/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3798311/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3798312/smarthack-wifi.log)

kueblc, 2019-11-01:
Do you have `python2` installed? Some reports suggest `psk-frontend` does not work with `python3`.

biotecov, 2019-11-02:
Good news and bad news:

The good news is that you were right, psk-frontend did not works as it should with python3 (I was innocently trying to rely only on python3 alone as the python2 developers have announced it will be deprecated). Getting back to python2.7, the intermediate firmware is installed. Navigating to http://10.42.42.42/ shows:

VTRUST-FLASH 1.2
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 414eb9
MAC: 50:02:91:41:4E:B9
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000

So thanks for the clue, which was correct.

The bad news is that the device became bricketed when I tryed to flash my alternative firmware.
Following the instructions, I copied a minimal 368K tasmota.bin to files/thirdparty.bin, and executed "curl http://10.42.42.42/flash3" from another terminal.
I then got the message:

"Device should flash http://10.42.42.1/files/thirdparty.bin and restart"

and from that point, my device never responded again by any means, neither through the fake tuya server nor through the tasmota encoded system  (I have no less than 20 devices with a variety of tasmota versions, so I am pretty used to).

Somehow, the intemediate firmware was sucessfullly flashed, bit not the the third-party firmware.
Is there anything I could try before I open and desolder the device (a really painful and risky task on this particular model) ? And, for the future, is there some extra precaution I should have take ?

Thanks for the tool and for the help, anyway,

biotecov, 2019-11-02:
Btw, here is the web log, 
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3800592/smarthack-web.log)
that show a line:

[I 191102 10:43:56 web:2246] 200 GET /files/thirdparty.bin (10.42.42.42) 7502.79ms

that seems to indicate that the thirdparty.bin firmware was loaded.
Further intents to connect using start_flash.bin were useless.

kueblc, 2019-11-03:
Can you share the firmware backup and the `thirdparty.bin` you tried to install? I will try to reproduce the issue here to see if I can find anything.

biotecov, 2019-11-04:
Here are the requested items. In the case of the thirdparty.bin it is a minimal tasmota firmware intended to allow flashing OTA regular tasmota firmwares. I compiled it from after hadrcoding some settings for my setup (I ardcodded my wifi SSID and password, my mqtt broker address, username and password, and the local co-ordinates). The version I send you has dummy values of exactly the same length and similar characters because I would not like to expose my true passwords and so on. Otherwise, they are identical to the flashed version.

This version works as expected when flashed using the serial connector, or when flashed OTA from a previous tasmota firmware.

I have recovered my device by flashing the firmware through the serial connector, so just for that unit, the job is done. However it remains that it would be nice if you could identify what prevents the the device from booting the tasmota firmware installed using the tuya-convert procedure for the future. I have ordered several extra units to install on other windows, and tuya-convert would be quit more convenient than the pain of opening this units. Besides, I guess It could be useful to others too.

Whether you can find something or not, thanks again for your work.

biotecov, 2019-11-04:
Well ... I am trying to upload the files, but the system says "We dont' support that file type".
So I compressed them.
[firmware-414eb9.bin.gz](https://github.com/ct-Open-Source/tuya-convert/files/3804586/firmware-414eb9.bin.gz)
[minimal.bin.gz](https://github.com/ct-Open-Source/tuya-convert/files/3804589/minimal.bin.gz)

kueblc, 2019-11-07:
I tried reproducing this issue using the supplied firmware backup. On bootup of the intermediate firmware I can see that it's throwing a system param error and dumping it to serial, but it did not prevent the bootup from completing. This makes sense, as you were able to get to `/flash3`.

This kind of issue can be observed when changing Espressif SDK versions, such as between the stock firmware and the intermediate. But I was unable to reproduce a bootloop or freeze by flashing the `minimal.bin` through `flash3`. Looks to have succeeded. I only ran into issues after running `undo`, which may also be due to the hardware.

One thing that did catch my eye was this note on startup of the `minimal.bin`:
```
00:00:00 APP: Set Baudrate to 19660500
```
Do you know if that is intentional/accurate?

biotecov, 2019-11-07:
I ignore why the initial baudrate is set to the max speed. I is not intentional, and after revising the code I cannot figure out how or why this happens. The only changes I made was to definie custom AP and MQTT servers and their passwords to speed up first connection. Otherwise I have compiled my custom tasmota-minimal.bin using the default values, the default platform and so on. But did not touch anything regarding serial speed. Maybe next time, when I receive the new units, I'll try to flash tasmota-basic.bin instead of tasmota-minimal.bin (maybe minimal lacks some essential startup procedures? well it is a blind guess, but repeating the same may give the same bad result, and tasmota-basic is still bellow 500K). Otherwise, instructions for manual flashing, through the serial port, recommend to erase the ESP before flashing the new firmware. I wonder if tuya-convert does that step and if that could be somehow related. Again a blind guess, sorry.

kueblc, 2019-11-07:
Yes, `tuya-convert` will erase the relevant parts of the flash. If started from `user2` it will erase `user1`, and vice versa. Upon flashing `user1` it will also erase the system params and the sectors used by Tasmota for settings.

0xFelix, 2019-11-09:
> Do you have `python2` installed? Some reports suggest `psk-frontend` does not work with `python3`.

I'd suggest to change the shebang of `psk-frontend.py` to python2 then. Distributions are beginning to default to python3. This error caught me.

See PR #382

kueblc, 2019-11-10:
Resolved with release v2.2.0, thanks for the report! Let us know if you run into any other issues.

closed 2019-11-10 by kueblc

biotecov, 2019-12-09:
@kueblc  Just to acknowledge your work: I finally received several devices like the curtain switch in this report, and I could flash tasmota using tuya-convert 2.3.0. The procedure is now painless and pretty fast! The same with two other brands of tuya-based switches (one had a secondary MCU, and the other adds 433MHz remote control). Warm thanks for your work!

kueblc, 2019-12-09:
@biotecov thank you, that's great to hear it worked out smoothly for you this time!