ct-Open-Source/tuya-convert #366: flash3 failed, LED blinking rapidly

Localhorst86, 2019-11-01:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?** No

**Do you have a question about configuring your newly flashed device?** No

**Did you have issues getting a device to flash?** YES

_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_
 Today I tried to convert my 4 existing Teckin SP21 Smart Plugs to tasmota using tuya-convert. This worked fine for 3 of my 4 devices. However, on the third device (chip ID 33c495) the flash3 seems to have failed. It flashed the intermediate firmware succesfully, made a backup and I was able to connect to the device using 10.42.42.42 and get the device info. When I started the flash3 process, the LED started flashing very rapidly and nothing further happened. I was not able to connect to 10.42.42.42 anymore and It didn't open its sonnof hotspot. Is my device bricked? Do I still have any chance of recovering it?
[device-info.txt](https://github.com/ct-Open-Source/tuya-convert/files/3798631/device-info.txt)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3798632/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3798633/smarthack-wifi.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3798634/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3798635/smarthack-mqtt.log)

kueblc, 2019-11-01:
Could you post your firmware backup? I can try to reproduce the issue

kueblc, 2019-11-01:
Looking at your web log, I don't see a request for `thirdparty.bin` around 33c495. Try running `start_flash.sh` again, this time you don't need to connect a phone or enter pairing mode, just unplug and plug the device back in.

Localhorst86, 2019-11-02:
ufortunately, your suggestion off unplugging and replugging didn't work.

What's interesting to note is the fact that the unit seems not to be completely dead, though. When I plug it in the outlet is off and the LED blinks rapidly, pressing the button on the plug once turns the unit on, but pressing it a second time doesn't turn it back off. The relay only turns off when I unplug the plug from the wall. Either way, I have uploaded the firmware backup.
[firmware-33c495.zip](https://github.com/ct-Open-Source/tuya-convert/files/3800607/firmware-33c495.zip)

Malfed, 2019-11-02:
Today I tried to flash my new Teckin SP21 and I got this problem, too. Like described in Localhorst86 first post I was able to locate the device using 10.42.42.42 and my final command was "curl http://10.42.42.42/flash3". After that the LED started flashing really rapidly and nothing further happend for about 25 minutes. I'm not able to connect to the device anymore.

In my case unplugging and replugging the device does not change anything. After plugging it in again it immediately starts flashing very rapidly again. The device neither does connect to my "vtrust-flash" wifi, nor does it pop up its own wifi network. Do you got any idea on how to recover it?

I've attached my logs and backup file, too. Maybe this will help.

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3801009/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3801010/smarthack-mqtt.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3801011/smarthack-web.log)
[firmware-5a7dbe.zip](https://github.com/ct-Open-Source/tuya-convert/files/3801012/firmware-5a7dbe.zip)
[device-info.txt](https://github.com/ct-Open-Source/tuya-convert/files/3801013/device-info.txt)

alekseyn1, 2019-11-06:
I had the same issue. "bricked" (hopefully not completely) 4 dimmers (out of 30). In all cases I had my own thirdparty.bin.  And in some cases tried to replace update.bin - with no luck. Perhaps that's what bricked them. Can not really identify what I did wrong exactly.  After the last one, recompiled Tasmota 6.4.1 (the last one that IMHO reliably works with tuya dimmer that I have) and replaced thirdparty.bin with it. All others reflashed just fine.

Will be patiently waiting for possible clues on how to solve it. I can pop those open and solder up the pins to reflash it, but it's a huge PITA (using Costco dimmers [Canada])

tre4b, 2019-11-06:
My Aoycocr plug did the same last night.  I used Flash3 aiming to put on the basic Tasmota.  As with above the light just flashes and the plug is now permanently on with the button doing nothing.  Turning on and off and trying to hold down the button does nothing.  Likelihood is it is now bricked, so I took it apart to see if it could be serial programmed.  Not sure I would reuse it though as my significant other would think it looked too risky ;-)

Question for me is it tells me to pay attention to the flash mode, however I could not figure out what that meant.  I got here looking for information about that.  I have watched videos of similar device flashes with the same Flashmode as mine though so that may not be the issue.

Localhorst86, 2019-11-06:
I used the included thirdparty.bin file.
I have now ordered a usb serial programmer and once that arrives (might take a while) I am going to try to flash the backup and start again, see if I can reproduce the issue - unless someone finds a way to recover the unit without opening it.

kueblc, 2019-11-06:
@alekseyn1 @tre4b if you did not use the included `thirdparty.bin`, did you make sure that the firmware fits within the 512KB limit?

Malfed, 2019-11-06:
@Localhorst86 If you manage to restore your device, could you perhaps give me a rough explanation of how you did it? The device was not that expensive, but I'd like to restore and use it anyway :)

Besides, I'd like to buy another device and try it again, but I'm still not sure what bricked my first device. In my opinion I followed all instructions and did use the included firmware.

kueblc, 2019-11-06:
@Localhorst86 @Malfed both of your logs do not show a request for `thirdparty.bin`. @alekseyn1 @tre4b did yours? Or was your `thirdparty.bin` larger than 512KB?

I believe I've isolated the issue and it is related to #362. In their case the firmware was too large. In your case the fetch for `thirdparty.bin` failed for some other reason, perhaps because the server was not running at the time `/flash3` was issued.

In both cases, after `/flash3` failed the device could not recover as it should. This is an issue with version 1.2 of the intermediate firmware. An update will be available by the end of the week that should resolve both #362 and #366.

Unfortunately if you've hit this issue your only recovery option may be to flash over serial. To be sure, check in your wifi log to see if the device connects to `vtrust-flash` after rebooting (unplug and replug device). If it does, you can still recover without serial.

alekseyn1, 2019-11-07:
Thanks  for your help @kueblc 
I can not recall now, but I believe I tried my best to keep it under 512K. At later stages I tried to keep it under 500K just to be safe.

tre4b, 2019-11-07:
@kueblc I just used the Tasmota one that came with Tuya-Convert.   I reasoned get this one on then update Tasmota using the OTA in Tasmota.  I didn't check the size as I assumed that it would be within the size settings.  I haven't checked the logs, will do so tonight and see what I can figure out.  Will also check the size of the included Tasmota.    I have no idea if the device is actually looking for the network so will check that too.

Localhorst86, 2019-11-07:
@kueblc I checked the WiFi logs, the bricked unit does not connect to the vtrust-flash network, I can only see a connection from my smartphone.
So I guess serial is the only way for me to recover, I just need to wait for the adapter to arrive from China.

kueblc, 2019-11-07:
@tre4b the included `thirdparty.bin` is certainly within the size limit. The download must have failed for some other reason.

tre4b, 2019-11-08:
Okay, the logs are not going to help.  When it failed I tried running it again without copying them so they have been overwritten :-(  Also how do I find out if the wifi is active.  I suspect not as having worked out the logs were not working I tried it again and it found nothing on the network to talk to.  Though I guess if it is in search for bin mode rather than see if someone wants to send an update mode that would be logical.  I'm running this on a Pi I set up to do this on last week so will be the most recent image.  How can I look at the network.

kueblc, 2019-11-08:
> When it failed I tried running it again without copying them so they have been overwritten

The logs are appended to, not overwritten. You should still have them unless explicitly deleted.

>  Also how do I find out if the wifi is active

Your `smarthack-wifi.log` will have an entry each time a device connects or disconnects, along with it's MAC address.

> Though I guess if it is in search for bin mode rather than see if someone wants to send an update mode that would be logical

Not sure what you mean. If you upload your logs I can take a look.

> I'm running this on a Pi I set up to do this on last week so will be the most recent image

Released v2.2.0 yesterday, update with `git pull` from the `tuya-convert` directory

tre4b, 2019-11-09:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3827783/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3827785/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3827786/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3827787/smarthack-wifi.log)

Okay logs as they stand here now.  I'm pretty sure the device is no longer working.  I have plugged it in and out a few times though so should show.  Not seeing anything but I may be reading the logs wrong.

I will give the next one in the box a go with v2.2.0 tomorrow.

kueblc, 2019-11-09:
As I suspected there's no request for `thirdparty.bin` in the web log. Unfortunately this means that `/flash3` failed and got into an unrecoverable state. This has been fixed in v2.2.0.

Since it doesn't look like it's reconnected in the wifi log, the only recovery option for this particular device is to flash manually over serial.

kueblc, 2019-11-10:
Resolved with release v2.2.0. Let us know if you run into any other issues.

closed 2019-11-10 by kueblc

tre4b, 2019-11-12:
@kueblc 

Quick questions.  How best to upgrade to v2.2.0?  Download and re-install or is there another way?

When done how can I check the installation is v2.2.0, just to make sure it is using new and does not brick another ;-)

kueblc, 2019-11-12:
If you followed the official installation instructions using git, just do `git pull`

tre4b, 2019-11-13:
I did.  Ta Colin will do

tre4b, 2019-11-13:
Just to confirm that this functioned perfectly this time with the v2.2.0 version.

Many thanks for your hard work.

kueblc, 2019-11-13:
Great, you're welcome!