ct-Open-Source/tuya-convert #367: install_prereq.sh fails without git

R0Wi, 2019-11-01:
I forgot to install git by mistake which leads to an error when executing `install_prereq.sh`.

```
Collecting git+https://github.com/M4dmartig4n/sslpsk.git
  Cloning https://github.com/M4dmartig4n/sslpsk.git to /tmp/pip-y3xpo6p8-build
  Error [Errno 2] No such file or directory: 'git' while executing command git clone -q https://github.com/M4dmartig4n/sslpsk.git /tmp/pip-y3xpo6p8-build
Cannot find command 'git'

```

kueblc, 2019-11-01:
You need `git` to install `tuya-convert` as described in the README

closed 2019-11-01 by kueblc

R0Wi, 2019-11-01:
Ok but what's wrong with adding it to the script?

kueblc, 2019-11-01:
If you're following [the installation instructions from the README](https://github.com/ct-Open-Source/tuya-convert#installation) you shouldn't be getting this error

R0Wi, 2019-11-08:
@kueblc i also noticed that in my debian distro the  `iw`-command was missing. So would it be good to add this package too?

kueblc, 2019-11-08:
Interesting, I thought this was a standard utility. What system are you running?

R0Wi, 2019-11-08:
Well i'm running OSMC on my raspberry pi  and i was too lazy to load a standard raspbian just for flashing tasmota :-) OSMC is currently based on Debian 9 but seems to lack the `iw`-package.

commit cee07b8d (2019-11-02): Merge pull request #368 from R0Wi/master

Add 'git' as dependency to fix #367
commit ee2f08f3 (2019-11-02): Add 'git' as dependency to fix #367