ct-Open-Source/tuya-convert #369: Hama RGB E14 176549 Device did not appear with the intermediate firmware

JoedowJoedow, 2019-11-01:
Hey *,
I tried to flash these Hama RGB bulbs but couldn't.

The script output looks that way:

pi@raspberrypi:~/tuya-convert $ sudo ./start_flash.sh
======================================================
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
....../start_flash.sh: Zeile 133:  3667 Beendet                 ./smartconfig/main.py
...........................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
Exiting...

I did the same one a switch KKmoon WiFi Smart Lichtschalter DIY Breaker Modul Smart Life/APP  and it worked, got SONOFF working there.

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3799899/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3799900/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3799901/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3799902/smarthack-web.log)

Maybe one of you can help me.
Thanks in advance
Greets

kueblc, 2019-11-01:
Might be related to #354

JoedowJoedow, 2019-11-04:
@kueblc: Thanks a lot for your fast reply. I will try to change the things - in fake-registration-server.py - as proposed. I will report soon if this was the problem.

JoedowJoedow, 2019-11-04:
> 
> 
> Might be related to #354

JoedowJoedow, 2019-11-04:
Hey "kueblc". The change in the fake-registration-server.py worked. Now I do have a "Sonoff/Tasmota" device. Once again thanks!!!

closed 2019-11-04 by JoedowJoedow

kueblc, 2019-11-04:
You're welcome, I'm glad it worked out!