ct-Open-Source/tuya-convert #373: After WLAN Connect with any device results in ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---"

Bobofet682, 2019-11-04:
I have the same issue as i found in issue #301:

carlzech described it:
I'm now trying it with a Raspberry Pi and a new WiFi-Stick. The WiFi does apppear but everytime I try to connect to the WiFi with my Smartphone the tool shows a lot of new lines and then the Raspberry Pi hang up. And I have to restart it because it's not responding to anything. I thought I made a mistake anywhere so I tried everthing again but I still have the same issue. The last two lines before it hang up: "Kernel panic - not syncing: Fatal exception in interrupt //// ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---" 
Can anyone help me with that?

_Originally posted by @carlzech in https://github.com/ct-Open-Source/tuya-convert/issues/301#issuecomment-544140830_

What I have tested:
As well Developement as Master Branch.
New installed and Updated (and upgraded) Raspbian Buster Lite.
3 different USB WLAN Sticks tested.
Raspberry 2 Model B V1.1 (As well tested with Raspberry 1 Modell B+)

What can i do to help fixing the problem?

Bobofet682, 2019-11-04:
sudo killall wpa_supplicant does not fix the Problem (was described in #301 as possible fix)
After Kernel panic occured, Raspberry wont responding to anything.
No Keyboard input, no ping, no ssh, nothing.

kueblc, 2019-11-04:
I don't think this is a `tuya-convert` bug. Did you check [Raspberry Pi](https://github.com/raspberrypi/linux/issues) and `hostapd` issues?

Malfed, 2019-11-04:
I just wanted to contribute that I encountered exactly the same issue on Saturday. Same as Bobofet682 I tried it again with both branches for several times, did the "killall wpa_supplicant"-thing (#301), but I always ran into the kernel panic, too. I didn't investigated it in greater detail so I don't know if it's an issue of Raspberry Pi, hostapd or tuya-convert, but it's always occuring when following the instructions of tuya-convert on my Raspberry Pi with Raspbian Buster Lite. My workaround was using a live usb image with Kali on my PC - maybe this might help you, Bobofet682.

kueblc, 2019-11-04:
`wpa_supplicant` is already taken care of in recent versions of `tuya-convert`, just to save anyone the trouble of trying that. Can you try running `hostapd` by itself and see if you can reproduce the issue?

Bobofet682, 2019-11-05:
ok, i do what i can....
i found an articla which describe:

new file: hostapd-test.conf
#change wlan0 to your wireless device
interface=wlan0
driver=nl80211
ssid=test
channel=1

sudo hostapd hostapd-test.conf

WLAN is activce, if i connect -> Kernel panic

Am i right if i say its a hostapd problem? How we can fix it for our case?

Bobofet682, 2019-11-05:
https://github.com/raspberrypi/linux/issues/2979#issuecomment-533995082

deleting the blacklisting-file /etc/modprobe.d/blacklist-rtl8192cu.conf did the job for my installation ... now the hostapd runs perfectly ...
sudo rm /etc/modprobe.d/blacklist-rtl8192cu.conf
now works :-)

kueblc, 2019-11-05:
Great find @Bobofet682, would be great to add an entry for this in the troubleshooting section on the wiki. Might be able to add an automated check for this in a future release of `tuya-convert` too.

kueblc, 2019-11-17:
Automated check staged for a future release. Feel free to test and provide feedback. Thanks for reporting and doing the research!

PR #401 (2019-11-19): Added check for blacklist file known to cause kernel panic in hostapd
Fixes #373
commit b74a708d (2019-11-19): Added check for blacklist file known to cause kernel panic in hostapd, fixes #373
closed 2019-11-17 by kueblc