ct-Open-Source/tuya-convert #375: Teckin SP23 keeps flashing rapid quick blue light after installing tasmota latest version

bachoo786, 2019-11-06:
### Before opening a new issue, please read the following
Hi 

I have a Teckin SP23 smart socket that I flashed using Tuya Over the Air. I followed everything correctly i.e. I had my phoned connected to "vtrust-flash" and was able to flash using my raspberry pi 2b+ connected via ethernet and a wifi dongle.

I also included the latest sonoff tasmota binary file and renamed it to thirdparty.bin. I then selected option 3 to install this tasmota binary file and it was successful.

However I cannot see the Tasmota AP on my phone or computer and the blue led power on/off light is flashing on my Teckin SP23 in a rapid manner. 

So I tried doing the whole process over again hoping I would be able to install a different binary file but my Teckin SP23 doesnt go in flash mode even if I turn it on or off, the blue led keeps flashing rapidly.

Here are my logs after trying to re do the process all over again:

smarthack-wifi.log:

```
lan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 24:18:1d:47:dd:be
wlan0: AP-DISABLED
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
Using interface wlan0 with hwaddr 00:0f:55:a8:d0:04 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 24:18:1d:47:dd:be
```

smartwifihack-mqtt.log:

```
^C1572999964: mosquitto version 1.4.10 terminating
1572999971: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1572999971: Using default config.
1572999971: Opening ipv4 listen socket on port 1883.
1572999971: Opening ipv6 listen socket on port 1883.
```
smartwifihack-web.log:
```

^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80
^[[32m[I 191106 00:26:15 web:2246]^[[m^O 302 GET /generate_204 (10.42.42.26) 5.69ms
^[[32m[I 191106 00:26:15 web:2246]^[[m^O 302 GET /generate_204 (10.42.42.26) 4.33ms
^[[32m[I 191106 00:27:16 web:2246]^[[m^O 302 GET /generate_204 (10.42.42.26) 5.61ms
```
smartwifihack-psk.log:
```
new client on port 443 from 10.42.42.26:54818
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54820
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54822
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54828
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54832
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 105, in <module>
    main()
  File "./psk-frontend.py", line 98, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
new client on port 443 from 10.42.42.26:54880
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54882
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54884
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54886
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54892
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54896
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54906
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54908
('could not establish sslpsk socket:', SSLError(1, u'[SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54910
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54914
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54916
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
new client on port 443 from 10.42.42.26:54922
```
the above error keeps continuing...

Can anyone please help me? 

Thanks.

kueblc, 2019-11-06:
Duplicate of #362. You tried to flash the latest Tasmota which is not within the limit (512kb). Unfortunately the only recovery option may be flashing over serial.

closed 2019-11-06 by kueblc

bachoo786, 2019-11-06:
@kueblc man I tried to flash another sp23 smart socket and same thing has happened. 

I installed the Sonoff-Tasmota v6.5.0  from your repository and after flashing I do not see the sonoff-XXX AP at all. 

The sp23 stays on forever cannot do anything to it. It is really frustrating that the thirdparty.bin file in your repository did not work!!!!!1

kueblc, 2019-11-06:
Unfortunately this is the inherent risk you take by flashing your devices, which is explicit in the README and also the first time you run `start_flash`.

Please be patient as we work to include additional safety checks to avoid situations like this in the future. I'd recommend waiting until our next release before trying it again, unless we can narrow down what happened in your case.

Do you wish you share your most recent web log and firmware backup so we can figure out what happened?

bachoo786, 2019-11-23:
Hey @kueblc so I now got the koogeek wifi smart plugs which are tuya compatible. 

I want to install tasmota in them before I break everything again. Can you please confirm that the tasmota file on your repo when I download it I can use it? It's just a standard basic tasmota bin file which I can upgrade later in the tasmota user interface?

And that the size of this tasmota bin in your repo is fine to flash on koogeek smart plugs?

Thanks I am holding fire until you tell me.

kueblc, 2019-11-23:
Yes @bachoo786 the included Tasmota has always been safe to use. Make sure to get the latest `tuya-convert` that includes the additional safety checks I mentioned before.

bachoo786, 2019-11-23:
Ok so I will download the files from the github and try it mate. Thanks

kueblc, 2019-11-23:
Use the install instructions on the README using `git`, this way you can update simply by doing `git pull`.

bachoo786, 2019-11-24:
ok I tried it and this is what I get after my socket is in pairing mode the light just stops blinking and it turns off.

```
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
...../start_flash.sh: line 106:  1687 Terminated              ./smartconfig/main.py
...........................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]
```

bachoo786, 2019-11-24:
@kueblc all sorted now working fine thanks alot for your help mate.

mahadi, 2019-11-26:
Same applies to me... what did you do to make it work?

bachoo786, 2019-11-26:
I was using raspbian stretch on my raspberry pi and once I installed raspbian buster as well as used koogeek smart plugs instead of teckin it worked fine