ct-Open-Source/tuya-convert #380: Logs seem to indicate device doesn't use ESP82XX when I have confirmed that it does

uatuba, 2019-11-07:
I am now attempting to flash a Teckin SP10. The log seems to indicate that this does not use the ESP82XX, but having torn one apart, I can assure you that it does. I'm not aware of the current firmware as I have not connected it to Tuya's servers and I do not know of another way to obtain that information. Any advice would be greatly appreciated:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3817757/smarthack-mqtt.log)

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3817760/smarthack-psk.log)

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3817762/smarthack-web.log)

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3817763/smarthack-wifi.log)

kueblc, 2019-11-07:
Is this from the one you tore apart? The same model can use different Tuya modules, the WR3 is designed to be pin out compatible.

fatfox511, 2019-11-07:
same issue here :(

I bought the LSC Smart Connect devices (LED Bulb, Motion Sensor and Door Sensor), all with live seen ESP82XX on board. 

I set them into pairing mode, they are blinking fast, after starting flash they stop blinking (Bulb for example has stativ light). Tuy tries resending smartconfig packets for a while then tells me " Device did not appear with the intermediate firmware", log indicates them as "non ESP82XX".

I just tried 2 different raspberrys, with different SD Cards with and without external wifi.

Is it normal, that there is no internet connection at the vtrust wifi?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3821202/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3821203/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3821204/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3821205/smarthack-wifi.log)

kueblc, 2019-11-07:
Are you able to retrieve a backup of the firmware over serial? I can use this to help debug the issue.

> Is it normal, that there is no internet connection at the vtrust wifi?

Yes

kueblc, 2019-11-07:
Also helpful would be a network capture of the device registering with the cloud through the app

fatfox511, 2019-11-09:
> Are you able to retrieve a backup of the firmware over serial? I can use this to help debug the issue.
> 
> > Is it normal, that there is no internet connection at the vtrust wifi?
> 
> Yes

I can try that next week!

The network capture is difficult for me as long as I do not want to use the app 😅 maybe I also give it a try.

oscar-rr, 2019-11-10:
same problem here. also saying that there is no ESP82XX

kueblc, 2019-11-10:
@oscar-rr have you opened the device to confirm?

kueblc, 2019-11-10:
Can any of you confirm this message is wrong with a picture of the ESP chip on the device that is giving this message?

This message was added to inform users why a flash might not be completing successful, but it does not alter the behavior of `tuya-convert` in any way. So if a flash is not completing with this message, removing the message will not make it more likely to complete.

fatfox511, 2019-11-10:
I can check again tomorrow and send a pic of proof, i will also try to do a backup via ftdi.

uatuba, 2019-11-10:
I threw away the device I destroyed to get to the chip, but it was absolutely an ESP chip. Are you saying that Tuya-convert doesn’t perform a check before flashing, but rather that his message is occurring for other reasons?

kueblc, 2019-11-10:
`tuya-convert` is not actively preventing you from flashing these devices. It is only printing a warning message to give you clues as to why it might not be working.

fatfox511, 2019-11-11:
[image]

This is a pic of the motionsensor. There is a TYWE3S on it. I think that should be able to flash, isn‘t it?

fatfox511, 2019-11-11:
[image]

And the doorsensor. Same chipset :)

meingraham, 2019-11-11:
Surely Tuya aren't labeling Wi-Fi modules that contain an ESP (e.g., TYWE3S) with the same model number as a module that contains a RTL chip? It seems that they'd have to differentiate the modules in FCC filings due to what is likely different Wi-Fi radio specs between the two chips.

fatfox511, 2019-11-11:
> 
> 
> Surely Tuya aren't labeling Wi-Fi modules that contain an ESP (e.g., TYWE3S) with the same model number as a module that contains a RTL chip? It seems that they'd have to differentiate the modules in FCC filings due to what is likely different Wi-Fi radio specs between the two chips.

Hmmmm I think you are right. I was fooled by the wifi Chip. There is a STML8 processor on the board. Maybe this one is doing the work.

https://www.mouser.com/datasheet/2/389/stm8l051f3-956454.pdf

In this case Tuya Convert doesn't seem to be the right choice for me ;)

What do the others think?

fatfox511, 2019-11-11:
Don`t wanna give up too fast:

Recording to this german forum https://www.mikrocontroller.net/topic/479796
the STM8 Chip is only for energy saving purpose and wakes up the ESP8266 within the TYW3S. One member claims that he was able to flash Tasmota on it but was not able to run it properly due to the energy savings controlled by the STM8 chip.

It seems that this one (STM8) works like a energy gatekeeper. Maybe this one should be bridged in order to wake up the ESP for flashing...

maybe something like this https://github.com/arendst/Tasmota/wiki/DM_WF_MDV4

kueblc, 2019-11-11:
Good work @fatfox511. This does add some depth to the mystery :) The next step would be getting a copy of the firmware so we can begin reverse engineering the differences.

@meingraham you would think, but the Teckin SP22 (#269) uses the same model number for both ESP and RTL based versions. Maybe this is an FCC violation, but IANAL.

fatfox511, 2019-11-11:
thank you @kueblc ! 

I stay tuned on the mikrocontroller forum asking the member how he's got an ftdi connection. 

There are some connecting plates (swim,rst, gnd, 3v3)  on the backside of the board, maybe I can gather some information before soldering. I don't know if I can use the swim "pin" and the rst for tx and rx.

If the worst comes to the worst i need to go directly on the TYWE3S Pins.

kueblc, 2019-11-11:
IIRC I think SWIM is a STM thing, "single wire interface module", not standard serial.

fatfox511, 2019-11-11:
[image]

Found the Solution on the backside :) soldering a littlebit now!

terryhonn, 2019-11-11:
I have no idea if the below is useful or not, I'm not a programmer, nor have I ever played one on TV.

I have a box of Teckin SP10s that I had hoped to put Tasmota on but couldn't get tuya-convert to connect to and flash. I tore one apart hoping to be able to flash via serial, but couldn't get to the location of the chip connections to solder without significantly destroying the enclosure, so I set them aside.  
After searching around again and finding this issue, I decided to unsolder the chip so that I could connect an FTDI adapter and see what happened. Below is the serial output from the chip from the time I grounded GPIO0 and applied 3.3v until after the tuya-convert connection attempts timed out.  If useful, fantastic. If not, please disregard and get back to useful activities. :-)

```
[01-01 00:00:49 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:49 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:50 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:50 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:50 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:50 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:51 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:51 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:51 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:52 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:52 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:52 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:53 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:53 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:53 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:53 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:54 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:54 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:54 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:55 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:55 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:55 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:56 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:56 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:56 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:57 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:57 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:57 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:57 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:58 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:58 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:58 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[... 3197 lines cut ...]

[01-01 00:00:24 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:24 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:24 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:25 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:25 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:25 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:26 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:26 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:26 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:27 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1

[01-01 00:00:27 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 0

[01-01 00:00:27 TUYA Notice][tuya_gpio.c:168] port = 23 ,level = 1
```

kueblc, 2019-11-11:
This is interesting data @terryhonn, thank you! It does confirm that in your case, the firmware is running on a RTL8195A.

fatfox511, 2019-11-11:
[pir.zip](https://github.com/ct-Open-Source/tuya-convert/files/3833467/pir.zip)

@kueblc after soldering and trying I was able to perform a backup. It seems to be an ESP8266EX with 1mb flash memory.

Maybe that helps :)

Now that I got constant power through my ftdi adapter I will try the tuya-convert method once again tomorrow :)

kueblc, 2019-11-12:
@fatfox511 it looks like this device, with `gwId 21020567ecfabc4bb81d`, failed to flash for a different reason. Looking back at your web logs I do not see the message
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```
But it does look like it's repeating the config calls, which has been indicative of schema problems, some of which may be resolved with release v2.2.0

kueblc, 2019-11-12:
The device producing the warning in your web logs was `uuid b1b803647e5aade4`

fatfox511, 2019-11-13:
Sorry for my lateness. Yeah thats right.

I tried the procedure multiple times with different devices (Door Sensor, Motion Sensor, Light Bulb), all of them are having the ESP in it, but maybe i mixed up the Log Files with on of the 3 devices.

The pictures of the PIR sensor and the backup bin file are just for an example and for further examination :)

I hope it helps anyway.

kueblc, 2019-11-14:
All evidence points to the warning message being accurate, and checking the SDK confirms. The [Tuya ESP8266 SDK](https://github.com/TuyaInc/tysdk_for_esp8266) contains no reference to `d.json` or `TUYA_IOT_SDK` while the [Tuya RTL8710 SDK](https://github.com/TuyaInc/tysdk_for_rtl8710bn/) and others contains no mention of `gw.json`.

Please feel free to post any new findings, but for now closing this issue as invalid since the message is presumably accurate.

closed 2019-11-14 by kueblc