ct-Open-Source/tuya-convert #381: Phillips WiZ bulbs

altimmons, 2019-11-09:
I bought these Phillips WiZ bulbs from Home Depot the other day. I tried to use them with the default software but it just doesn’t work well to integrate. It’s stuck on the cloud and it’s own app. 

I popped one open and took a look - expecting a big company like Phillips to have come up with a novel solution and yet sure enough inside was an ESP8266 WROOM-02D. 

Been trying to flash it though with little success. Only tried using the Tasmota and Espflasher tools. Not sure where Tuya fits into the picture but I was wondering if this is appropriate to use this on this product. 

The underlying developer of the light is TaoLight (rebranded to Phillips).  

I am hoping to get these to work as the light reproduction is of much higher quality than the standard bulbs that are reflashed. Many more LEDs and better ones. More worth the effort. There’s 3x the LEDs in general and 6x the White LEDs and the construction is of higher quality. 

Price was $12-$15 depending on the style. They have a vintage filament style (my wife) is particularly interested in.

kueblc, 2019-11-09:
Hi @altimmons 

These sound like quality bulbs, you've got me interested.

`tuya-convert` will only work on products with ESP based Tuya modules running on the Tuya cloud. I looked at the WiZ app just now, and it doesn't look like a Tuya based product. But it wouldn't hurt to try, if you find that they do pair with `tuya-convert` let me know!

closed 2019-11-09 by kueblc

divadiow, 2023-12-31:
did you get anywhere with a template? I've flashed this to Tasmota, it has an ESP-WROOM-02D

https://www.toolstation.com/4lite-wiz-led-a60-smart-bulb-wi-fi/p74038

I've got in functioning on/off, but no dimmer yet.

so far:

{"NAME":"4lite WiZ LED A60 Smart Bulb Wi-Fi 9W ES 806lm","GPIO":[0,0,0,0,224,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}