ct-Open-Source/tuya-convert #383: Cannot flash Cikuso PIR Motion Sensor

bruderflexo, 2019-11-10:
Cannot flash Cikuso PIR Motion Sensor (Amazon-Link:https://amzn.to/2Ct7OUv).
Don't know the firmware version.
I get this error when I try to flash then:

```
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 1s. ...............................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
```
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3828716/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3828717/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3828718/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3828719/smarthack-wifi.log)

I have no idea what I do wrong or if this device cannot be flashed.

kueblc, 2019-11-10:
It looks like you should be able to flash this device, but it's stuck on the config step. This might be a problem with the `schema`.

Are you able to retrieve either a firmware backup, registration capture, `schema`, `schemaId`, or `productKey` for this device?

bruderflexo, 2019-11-10:
I looked for some backup-files but I couldnt find anything. I didn't connect the device to tuya so I don't have everthings else either.

Where can I finde the `schema` or `schemaId` or `productkey`?

kueblc, 2019-11-10:
To get these you would need to record the registration with the Tuya cloud, or by registering and then setting up your device with `tuyapi`.

kueblc, 2019-11-21:
Closing while waiting for more information, please feel free to reopen when ready to follow up

closed 2019-11-21 by kueblc