ct-Open-Source/tuya-convert #384: Unable to flash Tuya WiFi motion Sensor

mycana, 2019-11-12:
The link to product page is [Tuya Smart WiFi Infrared Detectors Motion Sensor](https://www.aliexpress.com/item/33052022708.html?spm=a2g0o.productlist.0.0.5426652bU0BE0E&algo_pvid=be7a68e3-1d2c-4818-ba17-f86a03297844&algo_expid=be7a68e3-1d2c-4818-ba17-f86a03297844-1&btsid=d59bce32-b1e5-457f-a78d-6a22f116f27b&ws_ab_test=searchweb0_0,searchweb201602_1,searchweb201603_52)

[image]

Thank you.

kueblc, 2019-11-12:
Please post your logs

kueblc, 2019-11-12:
Related to #383?

mycana, 2019-11-12:
Log files:

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3834328/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3834329/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3834330/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3834331/smarthack-psk.log)

kueblc, 2019-11-17:
Looks like a duplicate of #135, try keeping the device active during flashing as suggested there.

kueblc, 2019-11-21:
Let us know if that worked for you or if you need further assistance. Closing the issue here, let's keep the discussion for sensors in #135.

closed 2019-11-21 by kueblc