ct-Open-Source/tuya-convert #386: SSL: NO_SHARED_CIPHER error when flashing Merkury A19

j2mc, 2019-11-12:
Two weeks ago I flashed a couple Merkury RGB bulbs with no problem, I'm now trying the A19 version and can't get past the initial connection.  

Logs:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3837090/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3837091/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3837092/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3837093/smarthack-wifi.log)

kueblc, 2019-11-12:
`SSL: NO_SHARED_CIPHER` is coming from your phone

kueblc, 2019-11-12:
Your device never connected, try the smart config again

j2mc, 2019-11-12:
Sorry, I should have noticed that in the logs.  Got it to work, ended up having to power cycle the bulb rapidly several times to get it to connect though.

closed 2019-11-12 by j2mc

kueblc, 2019-11-12:
No problem, glad you got it working