ct-Open-Source/tuya-convert #387: Question about a zigbee gateway?

ChristianPorsche, 2019-11-12:
Is there anything planned to convert a zigbee gateway, so it could be used directly in a different app for example. Smartlife and Tuya is working allright, but Apple Homekit, Google Home etc. is looking way better imo. Yes..google home is supported, but for almost every zigbee device not. 
Waiting for opinions if more people like to see a conversion for the zigbee gateway. I have this gateway right now: https://www.amazon.co.uk/TYGWZ-01-Gateway-Central-Controller-Devices/dp/B07N65MXD4

kueblc, 2019-11-12:
Have you tried? Does it have wifi?

meingraham, 2019-11-12:
I use openHAB interfacing to Tasmota devices for local and remote access via Google Assistant. I am just embarking on setting up the [Zigbee to Tasmota bridge](https://github.com/arendst/Tasmota/wiki/Zigbee) which should give me the ability to add Zigbee devices and expose them to openHAB through Tasmota.

kueblc, 2019-11-17:
Closing while waiting for clarification, feel free to reopen if this is still an issue

closed 2019-11-17 by kueblc