ct-Open-Source/tuya-convert #388: US DILISENS SP201 Power Monitoring Plug flashing issue

teslawave, 2019-11-13:
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3838815/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3838816/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3838817/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3838818/smarthack-mqtt.log)
having issues flashing the US DILISENS SP201 Power Monitoring Plug
the first time it started up connected and then when through a power cycle, i noticed when it was done though my phone had disconnected from the vtrust-flash, when i tried to flash again i keep getting these device did not appear with error.

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]

kueblc, 2019-11-13:
Can you clarify which device you're talking about?

Looks like device `c22d5da33a37feb7` can't be flashed:
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```

Looks like device `80006553807d3a7c7908` installed the intermediate flash loader, so it should be ready to install your custom firmware of choice. Just start the `vtrust-flash` AP and perform your desired `curl` commands.

teslawave, 2019-11-13:
sorry where do i find the device id # ?

kueblc, 2019-11-13:
In this latest run, nothing connected. Run the smart config again after putting the device back into pairing mode.

teslawave, 2019-11-13:
hey this is complete i figured out the issue, i don't know if this will be the same for everyone but i had to unplug the device several times until no lights on the plug were illuminated and without pushing  any buttons i tried to flash again and it worked.

closed 2019-11-13 by teslawave

kueblc, 2019-11-13:
Nice, you're welcome, glad you got it working.