ct-Open-Source/tuya-convert #390: Device did not appear with the intermediate firmware

barrymossel, 2019-11-13:
**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_
LSC Smartconnect light bulb, door sensor and movement detector, no firmware known
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3842909/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3842910/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3842911/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3842912/smarthack-psk.log)

Getting the following error:

```
pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 509.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 495.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 2s. ................................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
Exiting...

```

kueblc, 2019-11-13:
I only see two devices in the web log, but it looks like these two should be able to flash.

When it says "Do you want to try flashing another device? [y/N]", put the device back into pairing mode and then hit "Y". The `fake-registration-server` will try a different schema on the second try, which may resolve your issue.

barrymossel, 2019-11-13:
Unfortunately that didn't work. I tried three times in a row.

kueblc, 2019-11-13:
Could you share your most recent web log?

barrymossel, 2019-11-13:
Here you are:

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3843035/smarthack-web.log)

kueblc, 2019-11-13:
Would you try #393 and upload `smarthack-udp.log`?

barrymossel, 2019-11-14:
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3847014/smarthack-udp.log)

tidemann, 2019-11-15:
Having the same issue with two BlitzWolf BW-SHP2.
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3853395/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3853396/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3853397/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3853398/smarthack-psk.log)

kueblc, 2019-11-16:
@tidemann what makes you think this is the same issue?

tidemann, 2019-11-16:
@kueblc I got the same error message ("Device did not appear with the intermediate firmware"), and now they are dead.

barrymossel, 2019-11-16:
No news on this? I guess the udp log isn't very helpful...

kueblc, 2019-11-16:
@tidemann that message just means "it didn't work", you need to look in the logs to find out what the actual error is. Please open your own issue since we haven't tracked down the cause here and it doesn't seem to have anything in common with @barrymossel's logs.

@barrymossel sorry it looks like we couldn't capture the `productKey` via UDP broadcast. I'm looking into other options.

barrymossel, 2019-11-16:
If I can be of any help, please let me know!

kueblc, 2019-11-17:
@barrymossel just a hunch, but why don't you try changing `True` to `3` on [this line](https://github.com/ct-Open-Source/tuya-convert/blob/2cf2fa59ca04838da89b0c9ac273c1c4227676d3/scripts/fake-registration-server.py#L185).

kueblc, 2019-11-17:
You might also try changing `type` to `9`

kueblc, 2019-11-17:
Oops, scratch that, I was looking at @tidemann's logs. @tidemann you might try those steps.

@barrymossel for yours, can you try going into `fake-registration-server` and changing [this line](https://github.com/ct-Open-Source/tuya-convert/blob/2cf2fa59ca04838da89b0c9ac273c1c4227676d3/scripts/fake-registration-server.py#L149) from `20` to `10`?

My guess is that this is a `schema` issue. We either have to give it the correct `schema`, or give it one that keeps it from crashing long enough to trigger the upgrade.

If you have the true `schema` for this device, please do share it. You can obtain this by capturing a registration with the cloud, a firmware dump of the device, or if we manage to keep it online long enough it will broadcast UDP packets that we can use to determine the `productKey`.

barrymossel, 2019-11-18:
Changing it to 10 unfortunately didn't work. 

How would I capture a registration to the cloud? Or a firmware dump?

barrymossel, 2019-11-18:
By the way: quite weird actually that it happens for all my devices I bought last week (tried a E27 RGB bulb, a door sensor and a motion sensor). And these devices were reportedly flashable by tuya-convert (at least the bulb).

kueblc, 2019-11-18:
> How would I capture a registration to the cloud?

You would create a wifi bridge and record traffic with `tcpdump` or WireShark while registering the device using the app.

> Or a firmware dump?

You would need to disassemble the device and solder an FTDI to the chip and download the firmware using `esptool.py`. This is only if it is using an ESP82xx.

> (tried a E27 RGB bulb, a door sensor and a motion sensor). And these devices were reportedly flashable by tuya-convert (at least the bulb).

The [LSC E27 RGB is not compatible according to our wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTPS-firmware)) (see the note that it uses a different microcontroller). Door and motion sensors have their own issues since they use a separate microcontroller to power save, see #135 #355

barrymossel, 2019-11-20:
So I am out of luck with the bulbs? And would the sensors work if I flash them with a FTDI?

kueblc, 2019-11-20:
Yes unfortunately, the bulbs cannot run Tasmota etc. You might be able to flash the sensors if they use an ESP8266, but I'm not sure what firmware supports them due to their unique power saving.

kueblc, 2019-11-23:
Closing for now. Good luck, let me know if you need anything else.

closed 2019-11-23 by kueblc

barrymossel, 2019-11-23:
Thanks again.

rolandbreedveld, 2020-02-14:
Tried to flash the LSC Led strip, looks good, but after flashing the device is dead,
somebody any idea how to fix this ?

IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  66202      0  0:00:15  0:00:15 --:--:-- 32235
curl: Saved to filename 'firmware-c43baf.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: c43baf
MAC: A4:CF:12:C4:3B:AF
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v8.1.0.2 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...
Flashed http://10.42.42.1/files/tasmota.bin successfully in 8880ms, rebooting...
Look for a tasmota-xxxx SSID to which you can connect and configure
Be sure to configure your device for proper function!

HAVE FUN!