ct-Open-Source/tuya-convert #394: More Tecknic SP27 (UK) woes

ruffle-b, 2019-11-14:
Having sucessfully flashed some Teckin SP23s I bought a 4-pack of Teckin SP27s from Amazon.co.uk only to find that I can't flash them with tuya-convert :-(

I then found issue #364 (wish I'd spotted that before purchasing them) which seems to imply that these are just not usable.

Before I return them, I thought it might be worthwhile getting some more logs and info just in case it's a tweak to tuya-convert rather than a total fail.

Running tuya-convert, connecting a phone to the vtrust-flash AP and putting the Teckin into fast flash mode, I see the Teckin device get an IP allocated of 10.42.42.25 and it then broadcasts the following (captured with tcpdump -Xi wlan0 host 10.42.42.25):

```
09:02:33.776804 IP 10.42.42.25.49153 > 255.255.255.255.6667: UDP, length 172
        0x0000:  4500 00c8 0085 0000 ff11 865d 0a2a 2a19  E..........].**.
        0x0010:  ffff ffff c001 1a0b 00b4 a492 0000 55aa  ..............U.
        0x0020:  0000 0000 0000 0013 0000 009c 0000 0000  ................
        0x0030:  10ed e970 7e01 3aeb 94bb de01 6e18 38f8  ...p~.:.....n.8.
        0x0040:  d467 ca3a 334f 9dca aac2 75e2 a79d 451b  .g.:3O....u...E.
        0x0050:  4fdf f322 cc82 6461 a758 a161 8fce cd44  O.."..da.X.a...D
        0x0060:  09e2 035b 9c3e 1f16 19a8 8a35 3807 ed7f  ...[.>.....58...
        0x0070:  78da 9734 8091 5c56 bcd9 9fad 5a2f 88fa  x..4..\V....Z/..
        0x0080:  379e 7d68 a024 5626 9f46 1638 8e2a a04e  7.}h.$V&.F.8.*.N
        0x0090:  e667 ea0b 8f32 428e 7df4 72c4 054e ebe7  .g...2B.}.r..N..
        0x00a0:  fb79 4453 4fe5 989a 0c5b b393 4c9a c717  .yDSO....[..L...
        0x00b0:  3843 63dc d0f0 705a e4ca c595 bbf4 9625  8Cc...pZ.......%
        0x00c0:  6b1d d410 0000 aa55                      k......U
```

I'm guessing that tuya-convert doesn't like this as smarthack-web.log says:

```
POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=33&uuid=f0e10c184d9bd602&v=4.1&sign=e6257aa1e77f698e88d55f6907aedd4d
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK

payload 828653B6FFCAF268236DA6F62A762388
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"t":1573722052,"result":"DE5Knn9Cb3cqQPnnXE9wpt7q16ainqYPbWmWbMDljc/I8vUuAf86Ii7sZGk/V+R5cQF6B49KpXabHHGS1cxvxKqlAzvTue0k7+0uTuyd/yk+imXSpMR9ocIGrNaUeSSf","sign":"f96e9ba7d689a2d3"}
ESC[32m[I 191114 09:00:52 web:2246]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=33&uuid=f0e10c184d9bd602&v=4.1&sign=e6257aa1e77f698e88d55f6907aedd4d (10.42.42.1) 60.21ms
```

Are these really not ESP82XX based or are they some variant that's not recognised?  Taking one apart to look at the chip means breaking the glued-on casing which, as I have a 4-pack, I'm kind of reluctant to do as I won't be able to return them.

Here are the log files.

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3845642/smarthack-web.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3845643/smarthack-mqtt.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3845644/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3845645/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3845646/smarthack-wifi.log)

Happy to grab more logs and try stuff if it helps.

kueblc, 2019-11-14:
Yes unfortunately, it seems that this device is using an [RTL8710](https://github.com/TuyaInc/tysdk_for_rtl8710bn/). Installing firmware compiled for an ESP would brick the device and I'm not aware of any mature custom firmware for the RTL. Sorry for the trouble, hope you can exchange these for something more hackable!

closed 2019-11-14 by kueblc

ruffle-b, 2019-11-14:
Thanks Colin. They're on the way back to Amazon and I've sent an update to Blakadder's database to warn other folk.

tmiklas, 2019-11-25:
I can confirm the same issue with SP27

`WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
`

except mine don't want to work even with their original mobile app 👎 
that's definitely a return... may opt for an older SP23 or similar and try again

strowger, 2019-11-29:
The available items on Amazon UK are VERY QUICKLY switching to non-ESP82xx hardware.

I bought 20 https://www.amazon.co.uk/ANOOPSYCHE-Monitoring-Wireless-Sockets-Control/dp/B07PH28N1D from Amazon UK (with Prime next-day, rather than the cheaper "posted from China" option which seems likely to be non-ESP82xx) earlier this week and they were all ESP82xx and tuya-convert'able.

tmiklas, 2019-11-29:
Amazon UK product B07RYPJTJX is ok, runs perfect with `BlitzWolf SHP` template, flashed with `tuya-convert`

strowger, 2019-11-29:
> Amazon UK product B07RYPJTJX is ok, runs perfect with `BlitzWolf SHP` template, flashed with `tuya-convert`

Thanks!

I hate both square smartplugs, and smartplugs with the button on the side, though :-)

ruffle-b, 2019-11-29:
FWIW, I replaced the SP27's with Amazon UK item B07PHB9G2Y. Flashed fine.

Round with a button on the left so they won't please everyone :-) but at least they don't interfere with the socket switch like the SP23s do.

JonnyHaystack, 2019-12-15:
Can confirm I also got the B07PHB9G2Y Avatar plugs today and they're working great tuya-converted to Tasmota.

strowger, 2019-12-16:
Thanks for the suggestions. I'm still buying B07PH28N1D and as of a couple of days ago they're still working in tuya-convert.

Rustymage, 2019-12-18:
As of today, the order I recieved of B07QN5XY89 from Amazon do not flash.

`WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware`

strowger, 2019-12-19:
Sad times, but thanks for the warning - appreciated.

jmcharg, 2019-12-24:
Sorry to say I have just taken delivery of a pair of B07PH28N1D and they wont flash. So guessing they have the new chip in them as well now., They will be going back to Amazon this morning!

Rustymage, 2019-12-24:
I purchased 8 off B07PH28N1D (19th Dec) and so far, 4 have flashed fine. 

Have you check the smarthack-web log?

jmcharg, 2019-12-24:
> I purchased 8 off B07PH28N1D (19th Dec) and so far, 4 have flashed fine.
> Have you checked the smarthack-web log?

here is the log. nothing in there really tells me anything...

Received SIGINT, exiting...
Listening on 10.42.42.1:80
[I 191224 09:11:06 web:2246] 302 GET /generate_204 (10.42.42.35) 27.86ms
[I 191224 09:11:09 web:2246] 302 GET /generate_204 (10.42.42.35) 17.30ms
[I 191224 09:11:09 web:2246] 200 GET / (10.42.42.35) 23.86ms
[I 191224 09:11:09 web:2246] 302 GET /favicon.ico (10.42.42.35) 12.93ms
[I 191224 09:11:09 web:2246] 304 GET / (10.42.42.35) 16.66ms
[I 191224 09:11:10 web:2246] 302 GET /generate_204 (10.42.42.35) 17.05ms
[I 191224 09:11:10 web:2246] 302 GET /generate_204 (10.42.42.35) 15.15ms
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80
[I 191224 09:15:34 web:2246] 302 GET /generate_204 (10.42.42.35) 39.45ms
^CReceived SIGINT, exiting...

I have a dozen smart plugs mainly SP27's and about 60% have flashed successfully. I bought these B07PH28N1D yesterday to try to get a couple more and no luck this morning at all. If you've got any great ideas, I'd love to hear them :)

jmcharg, 2019-12-24:
Just a further update, I have double checked everything with the B07PH28N1D and they are running version 3.3 firmware, and will not flash. All the units I had that would flash were showing version 3.1 in the "webhack_udp.log" when they were seen on the network. I'll send these back to Amazon and try a different set. If I find ones that do work I will share on here again!

Rustymage, 2019-12-24:
I’m no expert but I’ll check my logs and see what they said. I remember when they didn’t flash it came up with a message saying they didn’t use ESPXXXX....

Did you have them with a fast flashing light and a device connected to the hotspot as it asks? Sorry if seems like I’m asking questions you’ve already checked.

My plugs clicked and then I had an error appear (line 106?) then after a 30/60 second wait it started to flash. When I flash another, I’ll post the logs up.