ct-Open-Source/tuya-convert #397: compatibility check, not listed here, listed in Tasmota templates repository

bartgrefte, 2019-11-16:
A while ago I got a smart plug that unfortunately depends on an internet connection, after having a look at the communication with an Android equivalent of Wireshark, it seems it's a Tuya smart plug.

This particular smart plug isn't listed on this github page as compatible, however, it is listed in the Tasmota templates repository -> https://github.com/blakadder/templates/blob/master/_templates/hIhome_WPP-16S1

I haven't tried to flash it yet, but since it's listed as compatible with Tasmota, does that also mean it's compatible with Tuya-convert?

kueblc, 2019-11-16:
Being compatible with Tasmota doesn't necessarily mean it's compatible with `tuya-convert`. It needs to be running Tuya firmware on an ESP82xx.

Also note that while a particular model may be listed as compatible on our wiki, they have recently started to change the microcontroller in these devices so it is hard to tell from the model alone. The only way to find out may just be to try it.

To end your session and close the `vtrust-flash` AP, run `stop_flash.sh`.

kueblc, 2019-11-16:
Let me know if you have any more questions and good luck!

closed 2019-11-16 by kueblc