ct-Open-Source/tuya-convert #398: setup_ap.sh logs error "unknow option 'i'"

R0Wi, 2019-11-16:
In the `scripts/setup_ap.sh` 
https://github.com/ct-Open-Source/tuya-convert/blob/2cf2fa59ca04838da89b0c9ac273c1c4227676d3/scripts/setup_ap.sh#L35
 there is a call to `hostapd` where the wifi interface option is given by the `-i` parameter. I'm using Debian 9.9 and getting the error `unknow option 'i'`. For me a quick fix was to directly manipulate the `scripts/hostapd.conf` by providing the correct wifi interface on my machine.
```bash
# interface file to control hostapd from tuya-convert
ctrl_interface=hostapd_ctrl

# our AP SSID
ssid=vtrust-flash

# manually specify channel, as ACS may not work on some systems
channel=1

# set log level to info to cut down on noise
logger_stdout_level=4

# added this line
interface=wlan0
```

Here's the output of my `hostapd -h` command:
```
hostapd v2.4
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2015, Jouni Malinen <j@w1.fi> and contributors

usage: hostapd [-hdBKtv] [-P <PID file>] [-e <entropy file>] \
         [-g <global ctrl_iface>] [-G <group>] \
         <configuration file(s)>

options:
   -h   show this usage
   -d   show more debug messages (-dd for even more)
   -B   run daemon in the background
   -e   entropy file
   -g   global control interface path
   -G   group for control interfaces
   -P   PID file
   -K   include key data in debug messages
   -f   log output to debug file instead of stdout
   -T = record to Linux tracing in addition to logging
        (records all messages regardless of debug verbosity)
   -t   include timestamps in some debug messages
   -v   show hostapd version

```

kueblc, 2019-11-16:
Duplicate of #358. Is `hostapd` v2.6 available for Debian 9.9?

kueblc, 2019-11-17:
I think that ideally we would specify the interface via command line option available in `hostapd` v2.6, but if it turns out v2.6 is not available for some users, we could go back to generating a config file on demand, or ask the user to edit `interface` in two locations.

R0Wi, 2019-11-18:
> Duplicate of #358. Is `hostapd` v2.6 available for Debian 9.9?

The official package sources for Debian 9 only provide `hostapd` version 2.4 (https://packages.debian.org/stretch/hostapd). 

> I think that ideally we would specify the interface via command line option available in `hostapd` v2.6, but if it turns out v2.6 is not available for some users, we could go back to generating a config file on demand, or ask the user to edit `interface` in two locations.

Well i also think it would be a good workaround to just copy the config file to a temporary one, add the one line and load it afterwards :-) If you're interested in this solution i would really like to contribute to this great piece of sotware.

kueblc, 2019-11-18:
OK sounds good, I would just like to avoid writing a file every time.

kueblc, 2019-11-19:
Resolved with #404, #405 

Thanks again for your contributions!

commit 3de90a86 (2019-11-19): Merge pull request #405 from R0Wi/master

Add backward compatibility for hostapd v2.5 and older (fix #398)
closed 2019-11-19 by kueblc