ct-Open-Source/tuya-convert #399: Cannot flash SCW NF101 switch

eku, 2019-11-17:
**Did you have issues getting a device to flash?**

I was not able to flash a [SCW NF101 switch](https://de.aliexpress.com/i/32963170549.htmll). It reacts on the Smart Wifi sequence, stopps flashing the LED and connects to the AP. That's all. Nothing more in the logs.

MAC is ```68:57:2d:13:12:be```.

[image]
[image]
[image]
[image]

kueblc, 2019-11-17:
Need more information. Can't help without your logs.

eku, 2019-11-17:
@kueblc as I wrote no logs at all beside the lines from starting the tuya-convert components.

kueblc, 2019-11-17:
Sorry I can't help much if I have nothing to go on. Try to make sure all the services started properly.

Are you sure this device contains an ESP82xx? I don't see one in the pictures you shared. I don't recognize the chip in the second picture.

kueblc, 2019-11-17:
[From the MAC address](https://aruljohn.com/mac/68572D) it does not look like an Espressif device, so we can't install ESP firmware.

eku, 2019-11-17:
Then I guess that's a bad buy. Thank you for your support.

closed 2019-11-17 by eku

kueblc, 2019-11-17:
You're welcome, sorry it didn't work out. Hope you can exchange these for something more hackable!