ct-Open-Source/tuya-convert #403: Bricked Teckin SB50?

123vollhorst, 2019-11-18:
Hi there,

I succesfully flashed some plugs (Teckin SP21) and some bulbs (Teckin SB50) with tuya convert and Tasmota. But two bulbs may have been bricked. They are just flashing blue (same freq when working ones are not connected) but there isnt any wifi interface nor the tuya app or tuya convert is recognizing them anymore. There isn just any connection to them anymore. Any ideas to get them running anyway? Logs are empty, too, so i wonder if I can trash them instead of spending so much time to get them to work.

Thanks

quan

kueblc, 2019-11-18:
If the bulb is flashing it is certainly not bricked. Sounds to me like it is waiting for you to connect to the `tasmota-xxxx` SSID to configure the wifi, or if you've done that already, it is waiting for you to configure the GPIOs.

It doesn't make any sense that the logs would be empty, if you ran `tuya-convert` you should at the very least see the services getting set up. Please share these logs so I can help further.

123vollhorst, 2019-11-18:
> It doesn't make any sense that the logs would be empty,

At least with the "old" tuya-convert logs were the same independent of a bulb on/off
I tried different notebooks/systems and checked router logs (but dunno what to look for in detail) and scanned network the whole time. Furthermore I tried all kind of (un)pairing procedures but the bulb didnt changed its behaviour in any way. 
Thanks
Kelvin

Logs:

 <details>
wifi

ttempting to stop wpa_supplicant
Configuring AP interface...
Error: Device for nexthop is not up.
Error: Device for nexthop is not up.
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:1e:09:31 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED a0:88:b4:60:e7:08

web

Listening on 10.42.42.1:80
^[[32m[I 191118 19:30:52 web:2246]^[[m^O 302 GET /static/hotspot.txt (10.42.42.32) 1.60ms
^[[32m[I 191118 19:30:52 web:2246]^[[m^O 302 GET /static/hotspot.txt (10.42.42.32) 2.98ms
^[[32m[I 191118 19:30:52 web:2246]^[[m^O 302 GET /static/hotspot.txt (10.42.42.32) 3.00ms
^[[32m[I 191118 19:30:52 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 0.87ms
^[[32m[I 191118 19:30:53 web:2246]^[[m^O 302 GET /static/hotspot.txt (10.42.42.32) 2.55ms
^[[32m[I 191118 19:30:53 web:2246]^[[m^O 200 GET / (10.42.42.32) 3.48ms
^[[32m[I 191118 19:30:53 web:2246]^[[m^O 302 GET /success.txt?ipv4 (10.42.42.32) 2.31ms
^[[32m[I 191118 19:30:53 web:2246]^[[m^O 200 GET /?ipv4= (10.42.42.32) 2.76ms
^[[32m[I 191118 19:30:55 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 1.02ms
^[[32m[I 191118 19:30:57 web:2246]^[[m^O 302 GET /static/hotspot.txt (10.42.42.32) 3.79ms
^[[32m[I 191118 19:30:58 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 2.20ms
^[[32m[I 191118 19:31:01 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 0.98ms
^[[32m[I 191118 19:31:04 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 0.81ms
^[[32m[I 191118 19:31:05 web:2246]^[[m^O 302 GET /static/hotspot.txt (10.42.42.32) 2.64ms
^[[32m[I 191118 19:31:07 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 0.95ms
^[[32m[I 191118 19:31:10 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 1.95ms
^[[32m[I 191118 19:31:13 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 1.74ms
^[[32m[I 191118 19:31:16 web:2246]^[[m^O 302 GET /success.txt (10.42.42.32) 0.88ms

mqtt

574099277: mosquitto version 1.5.7 starting
1574099277: Using default config.
1574099277: Opening ipv4 listen socket on port 1883.
1574099277: Opening ipv6 listen socket on port 1883.
^C1574099592: mosquitto version 1.5.7 terminating

psk

CTraceback (most recent call last):
  File "./psk-frontend.py", line 105, in <module>
    main()
  File "./psk-frontend.py", line 98, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
new client on port 443 from 10.42.42.32:45852
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45858
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45860
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45864
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45874
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45876
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45878
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45880
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45890
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.32:45896
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
</details>

kueblc, 2019-11-18:
It doesn't look like a device connected at all, so I don't see how you could have successfully flashed the plugs or bricked a bulb, unless you've cleared the logs since then.

123vollhorst, 2019-11-19:
Yes, there isnt any proof of existence as far as I can see, too. 
Its been some month since I flashed all devices, so I sadly dont have the logs anymore (Maybe with luck I can still find them on the device from then). I can remember that with one bulb I may have chosen the wrong firmware (flash3 or sth. like this with tuya convert) or sth. else noobish. The other one was handled like the working ones (Was found by TC, flashed, blinking and disappeared). 
Are there any other (amateur friendly) options/tools besides tuya-convert to get a possible connection to them? What the heck could have gone wrong here... :-/

kueblc, 2019-11-19:
> Yes, there isnt any proof of existence as far as I can see, too.
Its been some month since I flashed all devices, so I sadly dont have the logs anymore (Maybe with luck I can still find them on the device from then).

OK so if this is months ago, it could have been any number of bugs solved since then. Without the logs it is very difficult to tell.

> I can remember that with one bulb I may have chosen the wrong firmware (flash3 or sth. like this with tuya convert)

It won't let you use the wrong `flash2` or `flash3`, but you could have uploaded a bad firmware if you replaced `thirdparty` or instructed it to download a different one. It must be small enough to fit on the device and use the correct flash mode. But you said it blinks, so it is not bricked.

> Are there any other (amateur friendly) options/tools besides tuya-convert to get a possible connection to them?

The only other option is opening the device and flashing over wire.

Since you said it still blinks, it is running something. Are you sure no networks are created when you plug the device in? Does it try to connect to the `vtrust-flash` network?

> Edit: Got something I think. Didnt realize that without a bulb on the psk log is empty. So there is sth. that tries to communicate.... at least

Those are from your phone, not an IoT device.

kueblc, 2019-11-21:
Any updates @123vollhorst or is this resolved?

closed 2019-11-23 by kueblc