ct-Open-Source/tuya-convert #406: Cannot flash SONOFF T1 EU 2-gang wall switch

monsterflo, 2019-11-19:
Hello All,

I tried to flash the above module and ran into the SmartConfig loop.

Device is in Flashing mode (fast blinking LED).
Hardware: RaspberryPi 3+with Debian Stretch (2019-04-08)
I redownloaded the tuya-convert script last night.

logs are below.
Any help is highly appreciated.

Here is the console output:

```
root@raspberrypi:/tuya-convert# ./start_flash.sh
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fa                                                                            st). This is usually done by pressing and holding the primary button of the devi                                                                            ce
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
Exiting...

```
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3862631/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3862632/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3862633/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3862634/smarthack-web.log)

meingraham, 2019-11-19:
Because this device is NOT a Tuya device.

kueblc, 2019-11-19:
As @meingraham writes, this is a Sonoff device, not a Tuya device.

closed 2019-11-19 by kueblc