ct-Open-Source/tuya-convert #407: Failed flash of a Bn Link BNC-60/U133TJ

proludiojon, 2019-11-19:
Just got 4 of these, and 1st thing I did was try to convert one.
Before I started, I remove the entire tuya-convert directory, and got the clone on the latest, then ran the install_prereq.sh
It appears the plug is no longer working, as I cant turn on a light using the on-off button.
I also tried a brand new gosuna WP5 last night, and I think it did the same behavior, as that plug no longer works either.

[wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3863873/wifi.log)
[mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3863874/mqtt.log)
[psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3863875/psk.log)
[StartFlashCommandLog.txt](https://github.com/ct-Open-Source/tuya-convert/files/3863876/StartFlashCommandLog.txt)
[web.log](https://github.com/ct-Open-Source/tuya-convert/files/3863877/web.log)

kueblc, 2019-11-19:
I only see one device in the logs, `gwId 88412205c44f338d8b74`

This device appears to have installed the intermediate firmware. Restart the `start_flash` script, press enter (do not connect another device), and then plug your IoT device back in. Wait a second and it should pick up where it left off.

proludiojon, 2019-11-19:
That did the trick, and I also had to do the same on the other 3 I had.  Each one I had to restart.
My compliments on such a great product / hack.  Makes all our lives so much easier.

closed 2019-11-19 by proludiojon

kueblc, 2019-11-19:
Great! I'm glad that worked for you. Interesting that this was occurring for each of your devices, I'll consider this to see if we can improve the user experience here.

jaysalazar, 2020-07-25:
I am having similar experience as @proludiojon , however after restarting the start_flash script is states "Device did not appear with the intermediate firmware" message. The first WP5 was flashed without any issues, 2 and 3rd appear to be non functional now. I am   cautious to try the 4th and final one.

kueblc, 2020-07-25:
@jaysalazar your "non functional" devices are probably stuck on the intermediate firmware. Try steps in the Troubleshooting page on the wiki.

jaysalazar, 2020-07-26:
I looked over the troubleshooting page and unfortunately I do not see a "vtrust-recovery" SSID and it does not look like it is taking the 10.42.42.42 IP address. 
I do sense that it is not bricked, however I am not sure how to connect to the WP5 while it is under this state. I have tried started the IoT device and holding down the button on the side as well. 
I have also restarted the ./start_flash.sh and the PI itself, Ill see if I can pull the log files.

jaysalazar, 2020-07-28:
I was able to get all 4 plugs working after all, not 100% what I did or how many times I had to restart the plugs but I was able to talk to all of them eventually and configure the GPIO and MQTT. Thank you @kueblc for your help, I really appreciate it.

netstx, 2020-11-24:
In the comments [here](https://donnlee.com/2020/07/10/tuya-convert-bn-link-bnc-60-u133tj-wifi-smart-plugs/) looks like these devices may have been shipping with a new firmware (as of the last few months) that no longer allows the tuya-convert process to work. Has anyone tried recently?

erbfwa, 2020-12-21:
> In the comments [here](https://donnlee.com/2020/07/10/tuya-convert-bn-link-bnc-60-u133tj-wifi-smart-plugs/) looks like these devices may have been shipping with a new firmware (as of the last few months) that no longer allows the tuya-convert process to work. Has anyone tried recently?

I've been trying and failing with a batch I got last week. BNC-60/U133TJ

netstx, 2020-12-21:
> I've been trying and failing with a batch I got last week. BNC-60/U133TJ

Tried with 2x 4-packs and also couldn't get a single one flashed, so returned it to the store. I think these are done for...