ct-Open-Source/tuya-convert #409: Stuck at "Starting AP in a screen"

chrizbee, 2019-11-19:
Hey there,

I tried to run `start_flash.sh` but the script always gets stuck at `Starting AP in a screen`.
I am running a fresh Raspbian Buster Lite on a Raspberry Pi Zero W.

<details>
 <summary>smarthack-wifi.log</summary>
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:d5:5a:06 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED c0:ee:fb:d2:23:77
wlan0: AP-STA-DISCONNECTED c0:ee:fb:d2:23:77
wlan0: AP-STA-CONNECTED c0:ee:fb:d2:23:77
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:d5:5a:06 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED c0:ee:fb:d2:23:77
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:d5:5a:06 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:d5:5a:06 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
</details>

<details>
<summary>Log of `./start_flash.sh`</summary>
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 477.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] Y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 462.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen....
</details>

<details>
<summary>Log of `bash -x start_flash.sh`</summary>
++ tput bold
+ bold=''
++ tput sgr0
+ normal=''
++ cut -d . -f 2
++ screen --version
+ screen_minor=06
+ '[' 06 -gt 5 ']'
+ screen_with_log='sudo screen -L -Logfile'
+ . ./config.txt
++ WLAN=wlan0
++ AP=vtrust-flash
++ GATEWAY=10.42.42.1
+ ./stop_flash.sh
+ pushd scripts
+ . ./setup_checks.sh
++ . ../config.txt
+++ WLAN=wlan0
+++ AP=vtrust-flash
+++ GATEWAY=10.42.42.1
++ check_eula
++ '[' '!' -f eula_accepted ']'
++ check_config
++ iw list
++ grep -q '* AP'
++ echo -n 'Checking for network interface wlan0... '
Checking for network interface wlan0... ++ '[' -e /sys/class/net/wlan0 ']'
++ echo Found.
Found.
++ check_port udp 53 'resolve DNS queries'
++ protocol=udp
++ port=53
++ reason='resolve DNS queries'
++ echo -n 'Checking UDP port 53... '
Checking UDP port 53... +++ sudo ss -Hlnp -A udp 'sport = :53'
+++ head -n1
+++ grep -Po '(?<=pid=)(\d+)'
++ process_pid=477
++ '[' -n 477 ']'
+++ ps -p 477 -o comm=
++ process_name=dnsmasq
++ echo 'Occupied by dnsmasq with PID 477.'
Occupied by dnsmasq with PID 477.
++ echo 'Port 53 is needed to resolve DNS queries'
Port 53 is needed to resolve DNS queries
++ read -p 'Do you wish to terminate dnsmasq? [y/N] ' -n 1 -r
Do you wish to terminate dnsmasq? [y/N] y++ echo

++ [[ ! y =~ ^[Yy]$ ]]
+++ ps -p 477 -o unit=
+++ grep .service
+++ grep -Ev '^user'
++ service=dnsmasq.service
++ '[' -n dnsmasq.service ']'
++ echo 'Attempting to stop dnsmasq.service'
Attempting to stop dnsmasq.service
++ sudo systemctl stop dnsmasq.service
++ sleep 1
++ check_port udp 67 'offer DHCP leases'
++ protocol=udp
++ port=67
++ reason='offer DHCP leases'
++ echo -n 'Checking UDP port 67... '
Checking UDP port 67... +++ sudo ss -Hlnp -A udp 'sport = :67'
+++ grep -Po '(?<=pid=)(\d+)'
+++ head -n1
++ process_pid=
++ '[' -n '' ']'
++ echo Available.
Available.
++ check_port tcp 80 'answer HTTP requests'
++ protocol=tcp
++ port=80
++ reason='answer HTTP requests'
++ echo -n 'Checking TCP port 80... '
Checking TCP port 80... +++ grep -Po '(?<=pid=)(\d+)'
+++ head -n1
+++ sudo ss -Hlnp -A tcp 'sport = :80'
++ process_pid=
++ '[' -n '' ']'
++ echo Available.
Available.
++ check_port tcp 443 'answer HTTPS requests'
++ protocol=tcp
++ port=443
++ reason='answer HTTPS requests'
++ echo -n 'Checking TCP port 443... '
Checking TCP port 443... +++ sudo ss -Hlnp -A tcp 'sport = :443'
+++ head -n1
+++ grep -Po '(?<=pid=)(\d+)'
++ process_pid=
++ '[' -n '' ']'
++ echo Available.
Available.
++ check_port udp 6666 'detect unencrypted Tuya firmware'
++ protocol=udp
++ port=6666
++ reason='detect unencrypted Tuya firmware'
++ echo -n 'Checking UDP port 6666... '
Checking UDP port 6666... +++ head -n1
+++ sudo ss -Hlnp -A udp 'sport = :6666'
+++ grep -Po '(?<=pid=)(\d+)'
++ process_pid=
++ '[' -n '' ']'
++ echo Available.
Available.
++ check_port udp 6667 'detect encrypted Tuya firmware'
++ protocol=udp
++ port=6667
++ reason='detect encrypted Tuya firmware'
++ echo -n 'Checking UDP port 6667... '
Checking UDP port 6667... +++ grep -Po '(?<=pid=)(\d+)'
+++ sudo ss -Hlnp -A udp 'sport = :6667'
+++ head -n1
++ process_pid=
++ '[' -n '' ']'
++ echo Available.
Available.
++ check_port tcp 1883 'run MQTT'
++ protocol=tcp
++ port=1883
++ reason='run MQTT'
++ echo -n 'Checking TCP port 1883... '
Checking TCP port 1883... +++ sudo ss -Hlnp -A tcp 'sport = :1883'
+++ grep -Po '(?<=pid=)(\d+)'
+++ head -n1
++ process_pid=462
++ '[' -n 462 ']'
+++ ps -p 462 -o comm=
++ process_name=mosquitto
++ echo 'Occupied by mosquitto with PID 462.'
Occupied by mosquitto with PID 462.
++ echo 'Port 1883 is needed to run MQTT'
Port 1883 is needed to run MQTT
++ read -p 'Do you wish to terminate mosquitto? [y/N] ' -n 1 -r
Do you wish to terminate mosquitto? [y/N] y++ echo

++ [[ ! y =~ ^[Yy]$ ]]
+++ ps -p 462 -o unit=
+++ grep -Ev '^user'
+++ grep .service
++ service=mosquitto.service
++ '[' -n mosquitto.service ']'
++ echo 'Attempting to stop mosquitto.service'
Attempting to stop mosquitto.service
++ sudo systemctl stop mosquitto.service
++ sleep 1
++ check_port tcp 8886 'run MQTTS'
++ protocol=tcp
++ port=8886
++ reason='run MQTTS'
++ echo -n 'Checking TCP port 8886... '
Checking TCP port 8886... +++ sudo ss -Hlnp -A tcp 'sport = :8886'
+++ grep -Po '(?<=pid=)(\d+)'
+++ head -n1
++ process_pid=
++ '[' -n '' ']'
++ echo Available.
Available.
+ echo ======================================================
======================================================
+ echo -n '  Starting AP in a screen'
  Starting AP in a screen+ sudo screen -L -Logfile smarthack-wifi.log -S smarthack-wifi -m -d ./setup_ap.sh
+ ping -c 1 -W 1 -n 10.42.42.1
+ printf .
.+ ping -c 1 -W 1 -n 10.42.42.1
</details>

kueblc, 2019-11-19:
According to your wifi log, it does look like a station was able to connect. What is the IP and gateway assigned to the station? Does the `wlan0` interface get assigned `10.42.42.1`?

dooz127, 2019-11-20:
Seeing same issue running a nuc, kali 2019.4 docker container.

smarthack-wifi.log:

```
Stopping NetworkManager...
network-manager: unrecognized service
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp3s0...
Configuration file: hostapd.conf
Using interface wlp3s0 with hwaddr a0:c5:89:63:87:ba and ssid "vtrust-flash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
```

./start_flash.sh
```
root@nuc:/usr/src/tuya-convert# ./start_flash.sh 
Checking for network interface wlp3s0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.............................
```
ifconfig:
```
wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.42.42.1  netmask 255.255.255.0  broadcast 0.0.0.0
        ether a0:c5:89:63:87:ba  txqueuelen 1000  (Ethernet)
        RX packets 10  bytes 1872 (1.8 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 37  bytes 6378 (6.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

kueblc, 2019-11-20:
Maybe fixed with #351? Have you tried `development` branch or is this on `master`?

kueblc, 2019-11-20:
Ah never mind, #351 was merged directly into `master` so it shouldn't be that.

kueblc, 2019-11-20:
Would you see if this is perhaps a regression from #351? Try the commit directly before it, 8edc25129bd07a966a86417885f50882d218c70b

dooz127, 2019-11-20:
Same results but ifconfig:

```
wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.42.42.1  netmask 255.255.255.0  broadcast 0.0.0.0
        ether a0:c5:89:63:87:ba  txqueuelen 1000  (Ethernet)
        RX packets 219  bytes 32643 (31.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4530  bytes 606487 (592.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

chrizbee, 2019-11-20:
> Would you see if this is perhaps a regression from #351? Try the commit directly before it, [8edc251](https://github.com/ct-Open-Source/tuya-convert/commit/8edc25129bd07a966a86417885f50882d218c70b)

Does not work. Seems like the script gets stuck while ping'ing.
It's hard to check `wlan0` with `ip a`, since everytime I run `start_flash.sh` everything hangs.
I'm accessing the Pi with SSH and I have to reset the Pi manually (unplug, replug).

chrizbee, 2019-11-20:
I am now officially the stupidest person alive.
Of all my messing around I've forgot that I'm connected to the Pi with SSH...
No wonder it's losing its connection...
Sorry for wasting your time.

But `wlan0: Could not connect to kernel driver` might be a bug with `hostapd`.

closed 2019-11-20 by chrizbee

gimmel, 2022-07-19:
> I am now officially the stupidest person alive. Of all my messing around I've forgot that I'm connected to the Pi with SSH... No wonder it's losing its connection... Sorry for wasting your time.

That’s exactly what happened to me. I do everything on Raspberries via SSH. Had to buy a keyboard for flashing.
Thank you for pointing this out.