ct-Open-Source/tuya-convert #412: Enshine Bulbs: Device did not appear with the intermediate firmware

cwesterfield, 2019-11-21:
- Tuya Device
  - Enshine 60W bulb 2 pack https://amzn.to/2piEwoO
  - First bulb MAC ends with 88:EB
- Flashing Devices
  - Proxmox VM with edimax usb
  - Raspi 4 with built in wifi
  - Dell Laptop with internal wifi

I have attempted to flash these bulbs many times. I can get the bulb to connect to vtrust-flash, and pull a IP, but i always get the Device did not appear message. 

Once they connect to the AP, using fing I see the only open port I can see in 6668.

I have also attempted to press Y when I get the message and go right back into flashing (I think this changes something based on my research). 

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3872165/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3872166/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3872167/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3872168/smarthack-wifi.log)

cwesterfield, 2019-11-21:
The fine folks in the digiblur discord say that enshine has changed internals. These are almost definitely not esp8266 anymore.

closed 2019-11-21 by cwesterfield

kueblc, 2019-11-21:
Indeed the web log you shared says this much. Perhaps look into `tuyapi` as a way to control your Tuya device on the local network, and apply DNS rules to block cloud access.