ct-Open-Source/tuya-convert #413: TECKIN SP22 tasmota only makes backup.

Daviteusz, 2019-11-21:
I'm trying to flash tasmota to teckin sp22. Tuya-convert only makes firmware backup. I don't know what im doing wrong.

```
./start_flash.sh
Checking for network interface wlp3s0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 3472.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to terminate dnsmasq
kill: (3472): Nie ma takiego procesu
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
./start_flash.sh: linia 110:  4143 Zakończony             ./smartconfig/main.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  14968      0  0:01:10  0:01:10 --:--:-- 14365
curl: Saved to filename 'firmware-62c14c.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.3
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 62c14c
MAC: 84:0D:8E:62:C1:4C
BootVersion: 4
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A build of Tasmota v7.0.0.3 is already included in this repository.
   # curl http://10.42.42.42/flash3
If you want to flash the included ESPurna 1.13.5 image use this command:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/espurna-base.bin
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] y
======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the .. work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

....etc

Do you want to flash another device? [y/N] n
Exiting...
```

logs:
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3874311/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3874313/smarthack-wifi.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3874314/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3874315/smarthack-mqtt.log)

kueblc, 2019-11-21:
Did you follow the "Next steps"?

kueblc, 2019-11-21:
If you're looking for a more guided process that does everything automatically, check out the `development` branch. Otherwise you need to tell it what you want it to do.

Daviteusz, 2019-11-21:
> Did you follow the "Next steps"?
now i see. few months ago installation tasmota has been little bit different as now. I will try in 15 mins and report results.

Daviteusz, 2019-11-21:
Tasmota flashed succesfully. Closed.

closed 2019-11-21 by Daviteusz