ct-Open-Source/tuya-convert #414: Support for ESP8266_RTOS_SDK?

copercini, 2019-11-23:
Hi, first of all, thanks for the awesome work, this is very helpful

My question is, is it possible to make a thirdparty.bin using the https://github.com/espressif/ESP8266_RTOS_SDK structure?

It follows the same structure of ESP32 of flash partitions:
bootloader.bin: 0x1000
app1.bin (main program): 0x10000
partitions.bin: 0x8000

meingraham, 2019-11-23:
@copercini 

The third party firmware binaries packaged with Tuya-Convert are provided by other projects and not directly developed nor supported by this project (Tuya-Convert). If you are able to find a firmware binary with support for the parameters you listed, you can use that file instead as long as it is small enough and loads in the same flash region.

copercini, 2019-11-23:
@meingraham I understand, I tried to pack the ESP8266_RTOS_SDK hello world example in a thirdparty.bin file to flash using Tuya-Convert. It generates 3 files: 
bootloader.bin
partitions.bin
hello-world.bin
So I joined it with the proper offsets on a thirdparty.bin. But when trying to flash using Tuya-Convert, in the curl step, it just returns `error 4` or `error -1`. While flashing a known thirdparty.bin like tasmota it just works

Do you know what are these errors?

I'm attaching the files that I'm using for flash, maybe it helps somehow
[test_files_tuya_convert.zip](https://github.com/ct-Open-Source/tuya-convert/files/3882621/test_files_tuya_convert.zip)

copercini, 2019-11-23:
The exactly errors are:
` curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin
Flashing http://10.42.42.1/files/thirdparty.bin to userspace 1 failed after 53ms, error code 4`

`curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin
Flashing http://10.42.42.1/files/thirdparty.bin to userspace 1 failed after 17ms, error code -2`

`curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin
Flashing http://10.42.42.1/files/thirdparty.bin to userspace 1 failed after 1ms, error code -1`

copercini, 2019-11-23:
Oh, I found the intermediate firmware, this will help me a lot to understand : https://github.com/vtrust-de/esp8266-ota-flash-convert/blob/master/src/esp8266-ota-flash-convert.ino

copercini, 2019-11-23:
here it is:
```
HTTP server started
HTTP response Code: 200
HTTP response length: 252144
Magic byte from stream: 0xFF
Invalid magic byte
```
I'll adjust have to adjust my firmware to have it, or remove this check from esp8266-ota-flash-convert.ino

closed 2019-11-23 by copercini

kueblc, 2019-11-23:
@copercini that safety check is there for a reason, the correct solution is definitely not to cheat the check or remove it. Your offsets are wrong. See #389 for a safe way to assemble your `thirdparty.bin`.

copercini, 2019-11-23:
@kueblc thanks, setting the bootloader at 0x00, it worked =)

meingraham, 2019-11-23:
@kueblc 

These might make an interesting wiki article:
https://github.com/ct-Open-Source/tuya-convert/issues/389#issuecomment-553962100
https://github.com/ct-Open-Source/tuya-convert/issues/389#issuecomment-553972940

I'd offer to do it but I can confidently say that I wouldn't know what I was writing ;-)

Cheers!

Mike

kueblc, 2019-11-23:
Mike, good thought. See https://github.com/ct-Open-Source/tuya-convert/wiki/Flash-a-multipart-binary. Happy to have your input.