ct-Open-Source/tuya-convert #416: Any interest in making this cross-distro?

doenietzomoeilijk, 2019-11-24:
I have yet to determine whether it'll actually work (the bulb I bought turned out to be a dud, no flashing on switching it on, I'll swap it tomorrow), but I see no reason why Tuya Convert would only run on Debian (-based) distros. Personally, I'll be giving it a go on Fedora, and for completeness sake I could check OpenSUSE as well, as those are the distro's I run.

Would there be community effort in documenting it, and/or an attempt at making the scripts less directly dependant on Debian?

kueblc, 2019-11-24:
If it can be done in a way that doesn't significantly increase the maintenance burden, I'm all for it.

That said, it is not a priority. Accounting for all the differences possible in various environments is outside project scope.

Alternatively you are free to maintain a project for targeted to a specific environment. I believe there are Proxmox and Docker forks by community members.

kueblc, 2019-11-27:
Also see #233 #172

doenietzomoeilijk, 2019-12-16:
I think you're right about this not being a priority - supporting more distros means more work that doesn't go towards the actual flashing. As far as.I'm concerned, this issue can be closed, unless you want to keep it open for those.interested (which, seeing the activity on this issue, isn't a lot of.people 😃).

aenertia, 2019-12-23:
Why not use network manager - which is pretty cross distro (and these days actually usable) rather than wifi cli primitives which are a) hardware/distro specific and b) userspace dependant

iddo, 2020-02-02:
I've created this script which does everything specified in the instructions (except installing docker, and modifying the `config.txt` file), it should be pretty universal (I've succesfuly run this in Arch, which the instructions are not designed for).

```bash
#!/usr/bin/env bash
set -e

docker build --cache-from "tuya:latest" -t "tuya:latest" .

cleanup () {
  echo "========================="
  echo "Stopping docker container"
  docker stop "tuya"
}

trap cleanup EXIT

docker run \
  -d \
  --name="tuya" \
  --rm \
  --privileged=true \
  --network="host" \
  --volume="$(pwd)/backup:/usr/bin/tuya-convert/backups" \
  --env-file "config.txt" \
  "tuya:latest"

docker exec -it tuya start
```

Hope this helps

zarecor60, 2020-02-12:
I've got a fork of this using lsb_release to check the distro type, then branching to whichever startup script is required (only tested with Fedora at the moment, as it's the only one I have available right now). I can start a pull request if there's any interest. There weren't many changes outside the `install_prereq.sh` script.

aenertia, 2020-02-12:
There is the Universal Base Image (free as an beer and free as in speech)
rhel8 Image.

Upside being if you target it as the container image base you can if you
like get support for it.

I didn't rebasee the images but did a bit of work to make things work
nicely with the CNI tools (buildah,podman,skopeo) rather than docker.