ct-Open-Source/tuya-convert #418: Flashing Gosund SP1 fails: No cipher can be selected

chris1705, 2019-11-25:
Hello,

I'm trying to flash a Gosund SP1 socket with the help of tuya-convert. Unfortunately it always fails with following exception:

```('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))```

I have inspected the connection with the help of wireshark. The socket tries to establish a connection by using TLSv1.2 and lists TLS_PSK_WITH_AES_128_CBC_SHA256 as cipher suite, which seems to be the same as defined in psk-frontend.py.

Does anyone have an idea how to make it work?
Thanks!

Regards
Chris

kueblc, 2019-11-25:
Just to be sure, you have confirmed that this message is from the IoT device and not your connected smart phone?

chris1705, 2019-11-25:
Hi! Thanks for the quick answer. I'll get the same error for both devices:

```new client on port 443 from 10.42.42.1:49582
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.1:49584
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.16:45604
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.1:49586
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.1:49588
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.12:37907
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.1:49590
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.1:49592
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.1:49594
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.1:49596
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
```

I don't know why there are so many tries from 10.42.42.1 as this should by my own machine?
10.42.42.**16** seems to be my smartphone and 10.42.42.**12** seems to be the socket.

kueblc, 2019-11-27:
That `Client Hello` looks pretty normal to me. Do you see the server response?

kueblc, 2019-11-27:
Did the `install_prereq` go smoothly? Can you confirm your Python version and sslpsk install?

kueblc, 2019-11-27:
What is your OpenSSL version?

kueblc, 2019-11-28:
It may be that your OpenSSL is outdated as we found in #423

Boldie, 2019-11-28:
I have the same problem, installed a Kubuntu 19.10 to a flash drive, boot from it and got the same trouble.

kueblc, 2019-11-28:
Can you check your OpenSSL?

Boldie, 2019-11-28:
1.1.1c 28 May 2019

kueblc, 2019-11-28:
Should be OK there. Can you share your logs?

chris1705, 2019-11-29:
I retried with a fresh Ubuntu 19.04. LTS installation and everything worked like a charm.

The following pip packages were installed:
```asn1crypto==0.24.0
cryptography==2.1.4
enum34==1.1.6
idna==2.6
ipaddress==1.0.17
keyring==10.6.0
keyrings.alt==3.0
pycrypto==2.6.1
pygobject==3.26.1
pyxdg==0.25
SecretStorage==2.3.1
six==1.11.0
sslpsk==1.0.0```

OpenSSL: `1.1.1  11 Sep 2018`
Python 3 Version: `3.6.9`

Maybe those information will help someone.

Regards
Chris

Boldie, 2019-11-29:
[B806-B6E3.tar.gz](https://github.com/ct-Open-Source/tuya-convert/files/3905349/B806-B6E3.tar.gz)

So this is my log. This time made from an Xubuntu 18.04 but with no success. Maybe there is a different problem here. Maybe the firmware has been fixed or something else. My SP1 will stop blinking blue after starting the script but then nothing more happens. I will take the soldering iron now.

kueblc, 2019-11-29:
Excellent @chris1705 glad to hear it worked, and thanks for reporting back with more information!

@Boldie I see that the device connected but never made any requests. If you are going to flash over serial, would you please make a backup of the firmware to help debug this issue?

Boldie, 2019-12-01:
The requested file.
[fwbackup.bin.gz](https://github.com/ct-Open-Source/tuya-convert/files/3908867/fwbackup.bin.gz)

langerhans, 2019-12-03:
fwiw I just successfully flashed two Gosund SP1 with this script. They were reported as "latest version" (1.0.5) in the Smart Life app before I flashed them. Zero issues at all.

kueblc, 2020-01-25:
Thank you @Boldie for the very useful data. If you have any more devices to flash you can try with the latest release. Let me know if you still have any trouble.

closed 2020-01-25 by kueblc