ct-Open-Source/tuya-convert #419: Tried to revert to the original firmware on a DIN rail switch - ZMAI-90

teixeluis, 2019-11-26:
Was able to successfully convert this device to Tasmota. However,  as I expected, the metering features were not available. Before flashing I collected traces of the serial communication between the ESP8266 and the V8921 metering MCU, in an attempt to use the commands after the conversion, to retrieve the metering data. However, after reflashing, the V8921 does not respond to the commands. I assume that there must be initialization data sent during bootup which I was unable to collect.

In an attempt to put the device in the original state, I tried to use the backup generated by tuya-convert, and flash it using the NodeMCU PyFlasher. This resulted in the device entering an apparent loop condition, dumping what appears to be a repeating pattern of 21952 bytes through the serial port (at 115200 bps) - see attached file.

It there any obvious incorrection in the way the backup was reflashed? Here is the output of the command used for flashing (used NodeMCU PyFlasher):

```
Command: esptool.py --port COM3 --baud 115200 --after no_reset write_flash --flash_mode qio 0x00000 C:\Users\Luis\Documents\dev\home_automation\tasmota\zmai-90\firmware-xxxxxx.bin --erase-all

esptool.py v2.6
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.3s
Flash params set to 0x0030
Compressed 1048576 bytes to 509810...
Wrote 1048576 bytes (509810 compressed) at 0x00000000 in 45.2 seconds (effective 185.6 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
```
Is there eventually config data  (e.g. in the EEPROM) that the backup doesn't cover?

Thanks!

[bootup_mcu_to_v9821_with_backup_01_115200_seq.zip](https://github.com/ct-Open-Source/tuya-convert/files/3889366/bootup_mcu_to_v9821_with_backup_01_115200_seq.zip)

kueblc, 2019-11-26:
You should not need to manually set the flash mode with `esptool.py`. What is the original flash mode as reported by `device-info.txt`?

For the boot loop dump, try setting your baud rate to 74880, the bootloader default. 115200 is just going to produce garbage.

kueblc, 2019-11-26:
Just noticed also that `esptool.py` reported an auto-detected flash size of 2MB. `tuya-convert` has only been designed and tested with 1MB devices.

teixeluis, 2019-11-26:
Yes, device-info.txt has the following:

```
VTRUST-FLASH 1.3
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: xxxxxx
MAC: BC:DD:C2:xx:xx:xx
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000
```
Going to give another try, at 74880 as you suggested.

Thanks

Cheers.

teixeluis, 2019-11-26:
@kueblc regarding the flash size, this means that even if the firmware uses  more than 1 MB of the flash area, tuya-convert will truncate/backup only the (upper?) 1 MB of the flash?

I'll give it a try anyway. 

Cheers.

kueblc, 2019-11-26:
Did you get any more hints from the serial port at 74880?

The backup size is based on the firmware's reported size, not the real size of the flash chip. You should be able to confirm this, is your backup file 1MB? It's strange that they would use a 2MB flash chip and only use the first half of it.

teixeluis, 2019-11-26:
Hi @kueblc haven't yet tried. As soon as I get home will try it out. Yes the backup file is exactly 1 MB (1048576 bytes) in size.

Cheers,

Luis Teixeira

teixeluis, 2019-11-26:
Ok, tried again at 74880 bps and DIO instead. Result was the same. It keeps outputting garbage through the serial port. Flashed the tasmota firmware afterwards using the same settings, and it operates normally.

Settings used:

```
Command: esptool.py --port COM3 --baud 74880 --after no_reset write_flash --flash_mode dio 0x00000 C:\Users\Luis\Documents\dev\home_automation\tasmota\zmai-90\firmware-xxxxxx.bin --erase-all

esptool.py v2.6
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.7s
Flash params set to 0x0230
Compressed 1048576 bytes to 509810...
Wrote 1048576 bytes (509810 compressed) at 0x00000000 in 69.1 seconds (effective 121.4 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
```

There is a region in the beginning of the original firmware padded with 0's, which at a first glance is a bit curious:

```
$ head -c 512 firmware-xxxxxx.bin |hexdump -C
00000000  e9 03 00 20 7c 05 10 40  00 00 10 40 20 0a 00 00  |... |..@...@ ...|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000040  fc 82 fe 3f 1c 4b 00 40  cc 24 00 40 12 c1 e0 3d  |...?.K.@.$.@...=|
00000050  01 1c 04 02 61 07 01 fb  ff c0 00 00 32 01 00 22  |....a.......2.."|
00000060  a0 e9 27 13 1c 32 11 00  22 a4 ea 27 93 05 28 31  |..'..2.."..'..(1|
00000070  06 04 00 00 21 f3 ff 01  f4 ff c0 00 00 7c f2 46  |....!........|.F|
00000080  00 00 0c 02 08 71 12 c1  20 0d f0 00 aa 84 fe 3f  |.....q.. ......?|
00000090  08 fc 10 40 12 c1 f0 c2  61 02 20 c2 20 21 fb ff  |...@....a. . !..|
000000a0  c0 3c 20 09 31 01 e8 ff  c0 00 00 2d 0c c5 f9 ff  |.< .1......-....|
000000b0  26 02 14 31 f7 ff cc 32  2d 0c 06 01 00 c2 cc 10  |&..1...2-.......|
000000c0  2a 2c c0 03 00 46 00 00  0c 12 08 31 c8 21 12 c1  |*,...F.....1.!..|
000000d0  10 0d f0 00 00 10 00 00  14 10 00 00 18 10 00 00  |................|
000000e0  10 10 00 00 20 10 00 00  b4 18 00 40 00 4a 00 40  |.... ......@.J.@|
000000f0  4c 4a 00 40 91 fc ff 12  c1 e0 51 f7 ff 09 71 c9  |LJ.@......Q...q.|
00000100  61 d9 51 e9 41 f9 31 90  11 c0 3b f2 7d 03 c2 d1  |a.Q.A.1...;.}...|
00000110  10 1a 55 40 ef 11 79 05  dd 02 3d 0c 42 a0 04 e0  |..U@..y...=.B...|
00000120  2e 20 01 c8 ff c0 00 00  22 0c 00 0c 15 0c 26 8d  |. ......".....&.|
00000130  05 20 86 93 da 28 81 e9  ff 41 e6 ff 1a 88 59 08  |. ...(...A....Y.|
00000140  81 e8 ff 3d 01 10 88 80  62 68 00 40 22 11 01 bd  |...=....bh.@"...|
00000150  ff c0 00 00 31 e2 ff 81  e2 ff 1a 33 58 03 31 de  |....1......3X.1.|
00000160  ff 22 0c 00 1a 88 1a 33  68 08 78 03 cc 62 52 4c  |.".....3h.x..bRL|
00000170  00 5d 06 06 01 00 0c 02  22 4c 00 61 d8 ff 3d 07  |.]......"L.a..=.|
00000180  1a 66 52 66 00 0c 84 10  21 20 01 d7 ff c0 00 00  |.fRf....! ......|
00000190  81 d3 ff 1a 88 58 08 da  d5 2d 0d 01 d4 ff c0 00  |.....X...-......|
000001a0  00 41 cc ff 10 31 20 40  2d 11 01 d1 ff c0 00 00  |.A...1 @-.......|
000001b0  2d 0f 01 ce ff c0 00 00  2d 0e 3d 0c 0c 44 01 cc  |-.......-.=..D..|
000001c0  ff c0 00 00 91 c8 ff 9a  11 08 71 c8 61 d8 51 e8  |..........q.a.Q.|
000001d0  41 f8 31 12 c1 20 0d f0  04 09 00 60 00 09 00 60  |A.1.. .....`...`|
000001e0  20 4e 00 00 cc 2e 00 40  21 fc ff 12 c1 f0 09 31  | N.....@!......1|
000001f0  0c 03 c0 20 00 39 02 21  f9 ff 3c 83 c0 20 00 48  |... .9.!..<.. .H|
00000200
```

I think at this point I will buy a second device, and try to figure out the trickery that must be occuring during boot in the communication between the ESP and the v9821.

Thanks for the feedback.

Cheers,

Luis Teixeira

kueblc, 2019-11-26:
There's a bit of a misunderstanding. I did not mean to flash the device at 74880, but to read the boot messages at 74880. This might tell us why it's boot looping.

If you would like to share your firmware backup, I can try to see if there's any obvious corruption and try to run it myself.

teixeluis, 2019-11-26:
Ok @kueblc  sorry for the misunderstanding. Flashed again, and at the correct baud rate it makes much more sense :)

It keeps repeating this output:

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 16Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0.0(e8c5810) compiled @ Jan 25 2019 14:26:04
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 16Mbit(512KB+512KB)
jump to run user1 @ 1000
```

Cheers

kueblc, 2019-11-26:
That would explain it, the firmware is looking in the wrong place for the RF calibration data. Try explicitly setting flash size to 1MB with `esptool.py` when uploading.

teixeluis, 2019-11-26:
WiFi calibration data issue, maybe?

teixeluis, 2019-11-26:
Ok

teixeluis, 2019-11-26:
Ok, now it seems to be on the right path, will test with the Tuya app:

```

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user2 @ 81000

system param error
fd e7 ff ff ff ff ff ff 01 00 ff ff 0c 00 00 00 76 74 72 75 73 74 2d 66 6c 61 73 68 00 73 74 2d 
66 6c 61 73 68 22 2c 22 70 61 73 73 77 64 22 3a 01 01 00 00 00 00 01 00 75 6c 6c 2c 22 77 6b 5f 
6d 6f 64 65 22 3a 30 2c 22 6d 6f 64 65 22 3a 30 2c 22 74 79 70 65 22 3a 32 2c 22 73 6f 75 72 63 
65 22 3a 32 2c 22 70 61 74 68 22 3a 32 2c 22 74 69 6d 65 22 3a 31 2c 00 ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff 00 b8 27 eb da 35 f6 0a 00 00 00 45 53 50 5f 35 39 33 42 44 44 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 00 00 04 ff ff ff 01 00 ff ff 
0c 00 00 00 76 74 72 75 73 74 2d 66 6c 61 73 68 00 73 74 2d 66 6c 61 73 68 22 2c 22 70 61 73 73 
77 64 22 3a 00 75 6c 6c 2c 22 77 6b 5f 6d 6f 64 65 22 3a 30 2c 22 6d 6f 64 65 22 3a 30 2c 22 74 
79 70 65 22 3a 32 2c 22 73 6f 75 72 63 65 22 3a 32 2c 22 70 61 74 68 22 3a 32 2c 22 74 69 6d 65 
22 3a 31 2c ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 00 08 04 0c 11 1c 00 00 00 01 01 
ff ff ff ff 03 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff 64 00 ff ff 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff 
fc e7 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
[... 10 lines cut ...]
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
[... cut ...]
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff [06][00]™2™2™2™2™2™2™2™jRd™2™2™2™jR[04]‚[02]2™2™2™
…ªª[02]
…ªª[02]Š²[02]‚‚[02]‚‚[02]‚‚[02]Â[1A]‚¢[02]‚‚[02]‚‚[02]Â[1A]‚¢[02]‚‚[02]‚‚[02]Â¢[02]‚‚[02]‚‚[02]‚‚[02]*á[02]‚‚[02]‚‚[02]‚‚[02]j¤üÿC…«TRUST-FLASH 1.3
(c) VTRUST GMBH https://www.vtrust.de/35c3/
Connecting to Wifi.....................................................................................................
Failed to connect
Setting up SoftAP...
HTTP server started
```

kueblc, 2019-11-26:
Great! Since it looks like it's running the intermediate flash loader, you'll need to let it connect to the `vtrust-flash` AP and issue `curl http://10.42.42.42/undo` to reboot into the Tuya firmware.

teixeluis, 2019-11-27:
While moving on with the reverse engineering of the communication with the slave MCU (the v9821), I found that GPIO2 is being used to output serial data (soft UART I suppose). Set the bitrate to 74880, and found that it repeatedly sends:

```
[notice]mqtt_client.c:576 gw wifi stat is:5
scandone
no vtrust-flash found, reconnect after 1s
reconnect
scandone
no vtrust-flash found, reconnect after 1s
reconnect
[notice]mqtt_client.c:576 gw wifi stat is:5
scandone
no vtrust-flash found, reconnect after 1s
reconnect
scandone
no vtrust-flash found, reconnect after 1s
[notice]wf_sdk_adpt.c:1624 no_ap_cnt:60 and reconnect
[notice]mqtt_client.c:576 gw wifi stat is:5
scandone
no vtrust-flash found, reconnect after 1s
reconnect
scandone
```

Is this part of the tuya-convert routines?

Still trying to figure out what causes the v9821 to begin accepting the metering requests..

teixeluis, 2019-11-27:
Nevermind, just figured out that it is just debug output from the Tuya firmware (it prints the name of the AP it was last bound to - in this case the one from tuya-convert).

kueblc, 2019-11-27:
Seems like you've got the original firmware running again, anything else I can help with or can this be closed?

teixeluis, 2019-11-27:
Hi @kueblc, yes we can close this issue. After your guidance I was able to successfully revert to the original firmware (the main focus of this issue). Essentially it was a matter of flashing by forcing the flash-size to 1MB (had to run the esptool.py script manually instead of using the PyFlasher). Somehow the tool fails to identify the correct flash size for this particular chip, as it seems to be the case.

Thanks for your help.

Cheers,

Luis Teixeira

closed 2019-11-27 by teixeluis

CarlosGS, 2020-10-20:
> That would explain it, the firmware is looking in the wrong place for the RF calibration data. Try explicitly setting flash size to 1MB with `esptool.py` when uploading.

This was key to recover a TECKIN rgb strip. Memory was being auto-detected as 2MB and had to manually define it as 1MB. Thanks!! :)