ct-Open-Source/tuya-convert #423: Tuya curtain switch not flashing

Waldmensch1, 2019-11-27:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
no

**Do you have a question about configuring your newly flashed device?**
no

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_
There are no logs in the /scripts folder. Do I have to start the flash with additional parameters?

Device is (Printend on PCB) SC400W-EU-V2-Key   D-MFB190611. There is a TYWE3S soldered on.
I used tuya-convert a few weeks ago on 8 wallplugs which worked fine. So I assume that the environment (an Ubuntu Laptop) should be able to do the job. I pulled the latest code from github and runned the install_prereq.sh. Then I executed the start_flash.sh. I joined the vtrust WLAN with my iPhone and gave mains to the Tuya Device. One button shines solid. Then I long press the middle button until the LED starts flashing fast. Then I pressed return on the console to run flashing. After I see "Sending Secret" in the log the Device stops LED flashing to LED off.
In the log now is shown "SmartConfig complete" and "Resending SmartConfig Packets" a few times. After a while in log "Device did not appear....."

What I'm doing wrong? How can I help with logs

**Please remember to close your issue when the problem has been addressed!**

Waldmensch1, 2019-11-27:
Found a log "screenlog.0". I've shortened it because it contains "SSLError('No cipher can be selected.',)" hundreds of times.

`Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 00:21:85:bd:05:04 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Listening on 10.42.42.1:80
1574847917: mosquitto version 1.6.7 starting
1574847917: Using default config.
1574847917: Opening ipv4 listen socket on port 1883.
1574847917: Opening ipv6 listen socket on port 1883.
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
wlan0: AP-STA-CONNECTED 44:00:10:55:5a:2d
[32m[I 191127 10:45:47 web:2246][m 302 GET /hotspot-detect.html (10.42.42.13) 5.58ms
[32m[I 191127 10:45:47 web:2246][m 200 GET / (10.42.42.13) 6.57ms
[32m[I 191127 10:45:48 web:2246][m 302 GET /hotspot-detect.html (10.42.42.13) 3.78ms
[32m[I 191127 10:45:48 web:2246][m 200 GET / (10.42.42.13) 3.85ms
[32m[I 191127 10:45:48 web:2246][m 302 GET /hotspot-detect.html (10.42.42.13) 4.05ms
[32m[I 191127 10:45:48 web:2246][m 200 GET / (10.42.42.13) 4.60ms
wlan0: AP-STA-CONNECTED d8:bf:c0:54:50:99
new client on port 443 from 10.42.42.38:20526
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.38:11473
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.38:5550
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.38:21226
.
.
.
************ 100 times the same **********************
.
.
new client on port 443 from 10.42.42.13:61071
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.13:57322
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.13:57324
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.13:57326
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
wlan0: AP-STA-DISCONNECTED 44:00:10:55:5a:2d
wlan0: AP-STA-CONNECTED 44:00:10:55:5a:2d
[32m[I 191127 13:28:13 web:2246][m 302 GET /hotspot-detect.html (10.42.42.13) 5.51ms
[32m[I 191127 13:28:13 web:2246][m 200 GET / (10.42.42.13) 4.43ms
wlan0: AP-STA-DISCONNECTED 44:00:10:55:5a:2d
wlan0: AP-STA-CONNECTED 44:00:10:55:5a:2d
[32m[I 191127 13:34:27 web:2246][m 302 GET /hotspot-detect.html (10.42.42.13) 4.47ms
[32m[I 191127 13:34:27 web:2246][m 200 GET / (10.42.42.13) 5.26ms
wlan0: AP-STA-DISCONNECTED 44:00:10:55:5a:2d
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 104, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
^C1574858220: mosquitto version 1.6.7 terminating
^CReceived SIGINT, exiting...
^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED d8:bf:c0:54:50:99
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
`

kueblc, 2019-11-27:
The logs might be in your `backup` folder, but this screen log is helpful.

Can you perform a network capture?

To do this, make sure you have `tcpdump` installed, then run `start_flash`.

Wait until you get to the "Press ENTER to continue" prompt, then (without pressing enter) switch to another terminal and run `tcpdump -i wlan0 -w capture.pcap`.

Then plug your device in without resetting it (do not put it back in pairing mode, it should already be paired). Wait for a minute to collect data, press `Ctrl+C` to stop recording, and upload the `capture.pcap` file.

kueblc, 2019-11-27:
What is your OpenSSL version?

Waldmensch1, 2019-11-27:
Thanks for the fast response :)

`What is your OpenSSL version?`
```
openssl version -h
OpenSSL 1.0.2g 1 Mar 2016
```

The Dump is attached. I had to zip it to be accepted by git upload
[capture.zip](https://github.com/ct-Open-Source/tuya-convert/files/3897590/capture.zip)

kueblc, 2019-11-27:
I see a `Client Hello` but no `Server Hello`, indicating the problem indeed lies with OpenSSL. Is there a more recent version available for your distribution?

Waldmensch1, 2019-11-27:
Well, I set up a raspi 3B with tuya-convert and all prereqisites. Then I run a full cycle following the readme. With iphone connected to vtrust WLAN. It behaves same as on the Ubuntu Laptop. But on the raspi I got logs. I also had a tcpdump in a 2nd console. I put all in a zip and attached it to this comment. Maybe it helps identifying the issue. To be honest, I'm a greenhorn on this network magic :)

```
pi@raspberrypi:/home $ python --version
Python 2.7.16
pi@raspberrypi:/home $ openssl version -v
OpenSSL 1.1.1d  10 Sep 2019
pi@raspberrypi:/home $ tcpdump --version
tcpdump version 4.9.3
libpcap version 1.8.1
OpenSSL 1.1.1d  10 Sep 2019
pi@raspberrypi:/home $
```
[tuya-convert_logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/3898742/tuya-convert_logs.zip)

kueblc, 2019-11-27:
The device never connected this time. Did you put the device in pairing mode?

Waldmensch1, 2019-11-28:
Boar, call me stupid idiot. Havn't stopped the Laptop. So the devic may used this wifi instead of Raspi. So I stopped Laptop now and started Raspi unil "ENTER" Prompt. Then I powercycle the Device. Looking in the HTTP log say me that "things" are transmitted. I attached all logs in the zip again.
I did nothing further so far. So Raspi stays on ENTER Prompt and device is powerd witth mains. What are the next steps?
[tuya-convert_logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/3899820/tuya-convert_logs.zip)

kueblc, 2019-11-28:
Yes it looks like it took the intermediate firmware successfully! You should be able to continue now, just press ENTER.

Waldmensch1, 2019-11-28:
AWESOME!!!! So, learned from this: the new hack seems to not run with openSSL 1.0. (Ubuntu 16) The old Tuya wallplugs I was able to flash with. Maybe it helps other people.

Many thanks for your help, much appreciated.

```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  29704      0  0:00:35  0:00:35 --:--:-- 27850
curl: Saved to filename 'firmware-545099.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 545099
MAC: D8:BF:C0:54:50:99
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v7.0.0.3 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...
Flashed http://10.42.42.1/files/tasmota.bin successfully in 9896ms, rebooting...
Look for a tasmota-xxxx SSID to which you can connect and configure
Be sure to configure your device for proper function!

HAVE FUN!
======================================================
Do you want to flash another device? [y/N]

```

closed 2019-11-28 by Waldmensch1

Waldmensch1, 2019-11-28:
Just for Documentaion purposes 2 Pictures from Device.

- The Device has no brand as it looks. It's sold as "Smart Curtain Switch" for 10-15€ on [Aliexpress](https://de.aliexpress.com/item/32963064005.html?spm=a2g0s.9042311.0.0.67b14c4dDW8oRk)
- On PCB is printed **SC400W-EU-V2-Key** **D-MFB190611**. There is a TYWE3S soldered on
- Production date is end of Oct 2019 (see picture of the box)
- Serial flashing fails for whatever reason. Maybe RX/TX are wired somehow, which avoids communication. Also cutting RX connection as described on some Webpages doesn't help.
- The upper PCB can only detached by desoldering 6 pins which makes serial flashing hard anyway

[image]
[image]

imagined109, 2020-02-23:
@Waldmensch1 Hi - this may be a little off topic and I apologize for that but I'm new to this and really struggling.  I just got the same switch as you and I managed to flash tasmota (compiled with shutter).  However any and all configurations I try don't work - did you manage to get a working configuration going for this switch? Any help would be greatly appreciated.

MokanHan, 2020-06-25:
I received the same device yesterday SC400W but cannot OTA flash. Getting SSL errors:

`new client on port 443 from 10.42.42.40:10080
ID: 0242416f68626d64366147393149465231029018454923d84d6c89b646532ca05bcfccab21eadd53d068675cb89e8cd787
PSK: ea0c3221ce08e8b77fd348ecb1ded2681089ba77c239dbf495279ce84a286926
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:852)`

MokanHan, 2020-07-22:
> @Waldmensch1 Hi - this may be a little off topic and I apologize for that but I'm new to this and really struggling. I just got the same switch as you and I managed to flash tasmota (compiled with shutter). However any and all configurations I try don't work - did you manage to get a working configuration going for this switch? Any help would be greatly appreciated.
> 
> **EDIT**
> If anyone else stumbles upon this and comes across the same issue - this configuration ended up working for me: https://templates.blakadder.com/WF-CS01_EU.html

Hi were you able to flash with tuya-convert or did you flash serial? If Serial do you have a picture of the serial connections you made? I am am about to desolder the pcb and any info is appreciated.

imagined109, 2020-07-22:
> Hi were you able to flash with tuya-convert or did you flash serial? If Serial do you have a picture of the serial connections you made? I am am about to desolder the pcb and any info is appreciated.

Hi, I used tuya-convert, sorry. Also I should note that I ordered a second batch of these same switches and what came was not ESP-8266 based so I didn't flash those - loratap mentioned that they were switching over from esp to a different chip. Just an FYI

MokanHan, 2020-07-22:
> Hi, I used tuya-convert, sorry. Also I should note that I ordered a second batch of these same switches and what came was not ESP-8266 based so I didn't flash those - loratap mentioned that they [...]

Thanx. I am sure i have esp82xx based switch. Psk starts with 02. #483

MokanHan, 2020-08-07:
I flashed my devices serial. If interrested i can send backups.

pritam2016bakshi, 2023-03-07:
do you have backup firmware?